Proxmark/proxmark3 #875: Proxmark 3 does not support capturing Shanghai Fudan Microelectronics card data in China

ewangsoft, 2019-10-11:
Using “HF 14a snoop” command， to packetcapture，in proxmark3，Cannot capture card reader data，Card reader is the product of Fudan Microelectronics Company in Shanghai, China，The website is www.fmsh.com.
Only grab the data of the card.
It should be that proxmark 3 does not currently support it.

proxmark3> hf 14a snoop
#db# cancelled by button
#db# COMMAND FINISHED
#db# maxDataLen=3, Uart.state=0, Uart.len=0
#db# traceLen=205, Uart.output[0]=00000000
proxmark3> hf list 14a
Recorded Activity (TraceLen = 205 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error, ' denotes short bytes)            | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |      33536 | Tag | 0a  00  6f  15  84  0e  31  50  41  59  2e  53  59  53  2e  44  |     |
            |            |     | 44  46  30  31  a5  03  88  01  01  90  00  b7  2c              |  ok |
   35786160 |   35819632 | Tag | 0b  00  6f  15  84  0e  31  50  41  59  2e  53  59  53  2e  44  |     |
            |            |     | 44  46  30  31  a5  03  88  01  01  90  00  ff  7e              |  ok |
   63013776 |   63047312 | Tag | 0a  00  6f  15  84  0e  31  50  41  59  2e  53  59  53  2e  44  |     |
            |            |     | 44  46  30  31  a5  03  88  01  01  90  00  b7  2c              |  ok |
   91002144 |   91035616 | Tag | 0b  00  6f  15  84  0e  31  50  41  59  2e  53  59  53  2e  44  |     |
            |            |     | 44  46  30  31  a5  03  88  01  01  90  00  ff  7e              |  ok |
  142313792 |  142347328 | Tag | 0a  00  6f  15  84  0e  31  50  41  59  2e  53  59  53  2e  44  |     |
            |            |     | 44  46  30  31  a5  03  88  01  01  90  00  b7  2c              |  ok |
proxmark3>

pwpiwi, 2019-10-11:
I can think of two possible reasons for this behaviour:
1. you are using an RDV Easy which are infamous for their bad snooping performance.
2. the tag responses are quite long. Possible that the PM3 is still busy writing them to the trace buffer when the reader sends its next command.

ewangsoft, 2019-10-12:
> I can think of two possible reasons for this behaviour:
> 
> 1. you are using an RDV Easy which are infamous for their bad snooping performance.
> 2. the tag responses are quite long. Possible that the PM3 is still busy writing them to the trace buffer when the reader sends its next command.

I'm using PROXMARK3 EASY V3.0 of ELECHOUSE. Does this crawler have poor performance?
Is there any recommended hardware to capture 14443A data?

merlokk, 2019-10-12:
It needs to play with antenna's places reader and card. sometimes it better to place proxmark on one side of reader and card on another, sometimes needs to place it near card on its side, sometimes with space from card...
If nothing - needs to replace an antenna... handmade or buy separate [like this](https://lab401.com/collections/rfid-tools/products/proxmark-3-rdv4-long-range-hf-antenna1) or [this](https://www.aliexpress.com/item/32949403374.html) or i dont remember part number - i have one from molex with ferrite...

pwpiwi, 2019-10-12:
Regarding the RDV 3.0 Easy: I never had one on my desk. I am just referring to some posts on the Proxmark Forum. Buying another hardware might not help, if my second assumption would be true.

Try @merlokk 's suggestions. If they don't help, try another card type with shorter responses (e.g. a standard MIFARE card, if your reader supports it). If you can make it work then, my second assumption becomes much likely.

ewangsoft, 2019-10-13:
> It needs to play with antenna's places, reader's and card's. sometimes it better to place proxmark on one side of reader and card on another, sometimes needs to place it near card on its side, sometimes with space from card...
> If nothing - needs to replace an antenna... handmade or buy separate [like this](https://lab401.com/collections/rfid-tools/products/proxmark-3-rdv4-long-range-hf-antenna1) or [this](https://www.aliexpress.com/item/32949403374.html) or i dont remember part number - i have one from molex with ferrite...

Thank you. I tried. The problem remains.

pwpiwi, 2019-10-13:
Did you try with Mifare Classic card?

ewangsoft, 2019-10-13:
> Did you try with Mifare Classic card?

YES

pwpiwi, 2019-10-14:
Can you please post the result?

iceman1001, 2019-10-14:
Spare your time,  to answer your question proxmark doesn't read a fudan card like the one used in Shanghai.   The plan is to implement it one of these days.

iceman1001, 2019-10-14:
You can use raw commands to read out the first eight blocks of data.

pwpiwi, 2019-10-14:
He doesn't try to read the card, he tries to snoop a reader to card communication. If raw commands work, snoop should work as well.

ewangsoft, 2019-10-19:
Sorry, I forgot to pay attention to this problem recently
I've tried all the above suggestions, but I still can't catch the data of the card reader.

ewangsoft, 2019-10-19:
> You can use raw commands to read out the first eight blocks of data.

Sorry, I don't know how to operate the method you said. Could you give me some details? Thank you.

ewangsoft, 2019-10-19:
> Spare your time, to answer your question proxmark doesn't read a fudan card like the one used in Shanghai. The plan is to implement it one of these days.

Sorry, my poor English. Do you mean to support reading and writing Fudan card recently or something else?