Proxmark/proxmark3 #877: Broken after upgradig Proxmark3 to CDC Bootloader

bobby-2019, 2019-10-23:
After following instructions to upgrade Proxmark 3 to CDC Bootloader, I am not able to connect to the device it is not detectable. Both A and C red light on the Proxmark3 are permanently on when connected however Kali Linux is not able to detect the USB device. 
-https://github.com/Proxmark/proxmark3/wiki/Kali-Linux 

During the upgrade, I did not received any error as you can see below. Looking forward in receiving your advise, many thanks. 
[image]

pwpiwi, 2019-10-23:
LED A and C are switched on by the bootloader. So your Proxmark isn't broken and still starting the bootloader. But it cannot start the firmware.

Read about the modem manager issue in the Wiki and on the forum.

Then try the following: 
* disconnect the PM3
* press the button on the PM3 and keep it pressed.
* connect the PM3 to the USB port. Keep the button pressed.
* Check with `dmesg | grep -i usb` if your PC now recognises the PM3 and which USB-Port it uses. Keep the button pressed
* Flash the firmware (not the bootrom again). Keep the button pressed.
* Release the button after flashing is completed.

bobby-2019, 2019-10-24:
Many thanks for your assistance, PM3 is back online :)

closed 2019-10-24 by bobby-2019