Proxmark/proxmark3 #882: Compilation problem

LorenzoAlluminio, 2019-11-11:
Hi,
I'm trying to compile the source code on ubuntu 16.04 but the makefile fails with this error.
`iso15693.c: In function 'SetTag15693Uid':
iso15693.c:1695:2: error: missing braces around initializer [-Werror=missing-braces]
  uint8_t cmd[4][9] = {0x00};
  ^
iso15693.c:1695:2: error: (near initialization for 'cmd[0]') [-Werror=missing-braces]
cc1: all warnings being treated as errors
`

Any idea of why?

pwpiwi, 2019-11-11:
Which compiler version do you use? Please run
`arm-none-eabi-gcc --version`

LorenzoAlluminio, 2019-11-11:
This is the output:
`arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

pwpiwi, 2019-11-11:
Hmmm, prerelease anno 2015. I suggest that you upgrade to a newer version.

LorenzoAlluminio, 2019-11-11:
Updated, but now it fails like this:
`make[1]: uscita dalla directory "/home/allu/proxmark3/client"
make -C bootrom  all
make[1]: ingresso nella directory "/home/allu/proxmark3/bootrom"
arm-none-eabi-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
compilation terminated.
arm-none-eabi-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
compilation terminated.
arm-none-eabi-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
compilation terminated.
`

LorenzoAlluminio, 2019-11-11:
I tried to update it but I couldn't do it, it still gave me errors, but I managed to install correctly the Iceman fork. Thanks for the help anyway, really appreciated

closed 2019-11-11 by LorenzoAlluminio

LorenzoAlluminio, 2019-11-16:
I tried also with this fix and It solves my problem! Thanks!