Proxmark/proxmark3 #883: Support for 14443-4 APDU responses is limited to 512 bytes

darconeous, 2019-11-12:
Normal APDU responses have a maximum total size of 258 bytes: that's 256 bytes of data and a two-byte response code.

[Extended APDU](https://docs.oracle.com/javacard/3.0.5/prognotes/extended_apdu_format.htm) responses can have a maximum total size of 65538 bytes: 65536 bytes of data and a two-byte response code.

The proxmark3 firmware currently supports a maximum response size of 512 bytes, which is 510 bytes of data and a two-byte response code. This is normally adequate, but in some cases larger return values need to be handled.

For example, the [root CA certificate in a Tesla Model 3 Key Fob](https://gist.github.com/darconeous/2cd2de11148e3a75685940158bddf933#ins-0x06-get-certificate) is somewhat larger than 510 bytes. When requested, the entire certificate is included in the APDU response. There is no way to fetch subsets in order to build up the certificate over multiple APDU requests: the only way it is returned is as a large chunk.

Supporting the maximum theoretical 65538 byte length might not be practical, but 512 seems too small. Please consider doubling (1024) or quadrupling (2048) the maximum supported extended APDU length.

merlokk, 2019-11-12:
this size because proxmark catches signal then decodes. maybe it can be better. needs to look.
but) you need to use iso14443-4 chaining and all will be ok)

darconeous, 2019-11-12:
I had forgotten about chaining. I will look into that.

darconeous, 2019-11-13:
I managed to do a `hf list 14a` and extract the response manually. But still, this seems like something that the `apdu` command should be able to do without needing to manually piece together the response.