Proxmark/proxmark3 #885: About HF snoop

ewangsoft, 2019-11-15:
proxmark3> hf snoop
#db# Buffer cleared (40000 bytes)
#db# Skipping first 0 sample pairs, Skipping 0 triggers.

#db# Trigger kicked! Value: 255, Dumping Samples Hispeed now.
#db# HF Snoop end

Why can't I eavesdrop data

pwpiwi, 2019-11-15:
You probably did. Use 'data plot' to see the result.

ewangsoft, 2019-11-16:
> You probably did. Use 'data plot' to see the result.

Use 'data plot'，It's empty.

pwpiwi, 2019-11-18:
What do you mean with "empty"? Obviously there had been a trigger value. At least this one should be visible in 'data plot'.
Maybe it helps if you describe what you are trying to achieve, how you are trying to achieve it, and which commands you are using. And please post the results.

ewangsoft, 2019-11-18:
> What do you mean with "empty"? Obviously there had been a trigger value. At least this one should be visible in 'data plot'.
> Maybe it helps if you describe what you are trying to achieve, how you are trying to achieve it, and which commands you are using. And please post the results.

I used to snoop data between PCD and PICC with “hf 14a snoop", but I couldn't snoop PCD data. I thought it might be that PCD didn't conforming to 14443a standard , so I wanted to snoop PCD data with ”hf snoop",but I can't snoop the data.

I'm Chinese, I can't speak English. I translate it with translation software. I'm sorry.

pwpiwi, 2019-11-26:
If you see iso14443a PICC responses then definitely there had been iso14443a PCD requests. You need to change antenna position and distances until you see both PCD and PICC data after 'hf 14a snoop'.

closed 2020-02-13 by pwpiwi