Proxmark/proxmark3 #886: HITAG Fails on lf search

Conphilpott, 2019-11-16:
Hitag used to work a few days ago now it just fails very time I try to do a lf search or run any lf hitag commands.

Prox/RFID mark3 RFID instrument
bootrom: HEAD/v3.1.0-suspect 2019-08-19 06:53:30
os: HEAD/v3.1.0-suspect 2018-12-01 08:57:42
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2018/09/12 at 15:18:46
SmartCard Slot: not available

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 194091 bytes (37%). Free: 330197 bytes (63%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
proxmark3> lf search
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

#: DEBUG: Error - hitag failed

No Data Found! - maybe not an LF tag?

proxmark3>

pwpiwi, 2019-11-21:
You may try PR #887. This eliminates the Error message during 'lf search' and adds some reliability to Hitag functions in general.

On RDV4 with the standard antennas Hitag functions fail most of the time.

Please note that 'lf search' only detects Hitag2 but not HitagS.

pwpiwi, 2019-11-29:
#887 is now merged to master. Just in case you don't know how to apply patches...

cosmo6110, 2020-03-04:
I test your latest version on proxmark3 easy and rdv4.
On the rdv4 i used the ex range antenna. The search function didnt work on both device.
Tried the v3.1.0 and it's worked, but the tag must be in a exact position against the proxmark3.

v3.1.0

closed 2025-10-14 by conphilpott