Proxmark/proxmark3 #891: Bricked Proxmark3 rdv4 - can't fix!

jp50pr, 2019-12-03:
I have a Proxmark3 rdv 4 I bricked...well because I have no idea what I am doing. I purchased a j-link EDU, I also purchased the lab 401 adapter. I thought I followed the guide on how to fix it but I keep getting an error.
Any assistance would be greatly appreciated. 

[image]

merlokk, 2019-12-03:
Looks like you don't connect jtag correctly. One of errors - it can't see vcc voltage

iceman1001, 2019-12-03:
Did you follow these instructions?  https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/jtag_notes.md   I see your jtag configured for AT91SAM7S256 ,  where RDV4 uses AT91SAM7S512

pwpiwi, 2019-12-03:
> I have no idea what I am doing

In such a case you should stop and breath slowly instead of connecting wires randomly. What makes you think that your proxmark is bricked? What do you observe when you connect it to your PC? LEDs?

jp50pr, 2019-12-03:
Sorry I could not get it all in the picture. The reason I think it is bricked it because my system does not recognize it and all lights turn on when there is power.

I am using the adapter to "avoid" wiring issues.

[image]

jp50pr, 2019-12-03:
I used https://github.com/Proxmark/proxmark3/wiki/De-Bricking-Segger for the guide, which is where I saw "exec device = AT91SAM7S256". I did just try using "exec device = AT91SAM7S512" but no luck still.

iceman1001, 2019-12-03:
I recommend you use the jtag instructions I linked.

jp50pr, 2019-12-03:
Thank you all for the help and responses. Iceman1001, I used your recommendation and used the guide provided. I appreciate the help. Now that it is no longer bricked time to get this guy working again.

closed 2019-12-03 by jp50pr