Proxmark/proxmark3 #892: My proxmark3 is temperature sensitive,warm make it work.

SDPatrick, 2019-12-03:
Hi  guys, My proxmark3 had a very wired problem.
works fine at high temperature and failed at low temperature.
**1:**  
hardware info~
~~~~~~~~~~~~~~~~~~~~~~~
pm3 --> hw version          
Proxmark3 RFID instrument
[ ARM ]
 bootrom: iceman/master/ice_v3.1.0-1094-g97128ad5 2019-08-28 01:58:55
      os: iceman/master/ice_v3.1.0-1094-g97128ad5 2019-08-28 01:59:00
[ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          
 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev A          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 235688 bytes (45%) Free: 288600 bytes (55%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
**2:**  
can’t read all HF tags when it’s cold (eg. winter).
eg. pm3 --> hw tune 
[=] measuring antenna characteristics, please wait...
[!] LF antenna is UNUSABLE 
[+] HF antenna: 30.23 V - 13.56 MHz          
[+] HF antenna is OK      
[-] Not showing LF tuning graph since all values is zero. 
pm3 --> hf 14a info          
[!] iso14443a card select failed          
    
**3:**
works normal after I heated the PCB board
pm3 --> hw tune  
[=] measuring antenna characteristics, please wait...
 [!] LF antenna is UNUSABLE 
[+] HF antenna: 29.83 V - 13.56 MHz          
[+] HF antenna is OK     
[-] Not showing LF tuning graph since all values is zero. 
pm3 --> hf 14a info          
 UID : 84 17 25 02           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1          
[=] proprietary non iso14443-4 card found, RATS not supported          
[+] Answers to magic commands (GEN 1a): YES          
[+] Prng detection: WEAK          
pm3 -->           

noted:
removed the LF antenna and tags tested at same postion.
all the situation above can be reappearance from cold to hot
already use the heater air gun to rewelding all the component. the problem still reappearance.

Can any folks give me some advice or suggestions?
I suspect that Capacitor ruined…. willing replaced it…

to be continued..

thanks for your idea .

pwpiwi, 2019-12-03:
How cold is "cold" and how hot is "heated"?

iceman1001, 2019-12-03:
Which Proxmark3 device do you have?

SDPatrick, 2019-12-04:
> How cold is "cold" and how hot is "heated"?
 down under approximately 120.F its not work.
step 3 heated means approximately 150F or higher
here is the thermal image when it "cold" and connected to pc
!20191204085645](https://user-images.githubusercontent.com/58476865/70102774-1f938380-1674-11ea-977e-603013d55b2b.jpg)

SDPatrick, 2019-12-04:
> Which Proxmark3 device do you have?
a chinese modification called proxmark easy 3.with the hf antenna on pcb and a lf antenna removable.
 I supposed that it's a cheap version of PM3 original.

SDPatrick, 2019-12-04:
> Which Proxmark3 device do you have?
a chinese modification called proxmark easy 3.with the hf antenna on pcb and a lf antenna removable.
I supposed that it's a cheap version of PM3 origin

SDPatrick, 2019-12-04:
> Which Proxmark3 device do you have?
@iceman1001 
a chinese modification called proxmark easy 3.with the hf antenna on pcb and a lf antenna removable.
I supposed that it's a cheap version of PM3 original

SDPatrick, 2019-12-04:
replaced  2 big capacitor ,still failed
replaced switch 74HC4066,still failed
replaced cysrtal 13.56mhz,still failed

iceman1001, 2019-12-04:
hm,  I see you use iceman fork,  how about you try the [RRG/Iceman repo](https://github.com/rfidresearchgroup/proxmark3)  and make sure you compile it according to this document [PM3OTHER](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md)  

Then I would like to see the output from 
-hw status
-hw tune
-hf 14a info

SDPatrick, 2019-12-04:
> hm, I see you use iceman fork, how about you try the [RRG/Iceman repo](https://github.com/rfidresearchgroup/proxmark3) and make sure you compile it according to this document [...]

OK,I'll try it weekend ,tks.
you tought it's a software problem instead of hardware problem?why? any clues?

iceman1001, 2019-12-04:
No control over the different manufactures of Proxmark3 hardware.  Its totally different.
On the forum, you see that we try to identify problematic versions from sources.
A clone of the Pm3 v30 Easy is notorious for some hw issues.  However not related to heat/cold.

Some stuff we identified and corrected with fw,  which is why I want you to test.

SDPatrick, 2019-12-04:
> No control over the different manufactures of Proxmark3 hardware. Its totally different.
> On the forum, you see that we try to identify problematic versions from sources.
> A clone of the Pm3 v30 Easy [...]

OK，I‘ll test RRG repo asap

SDPatrick, 2019-12-06:
> No control over the different manufactures of Proxmark3 hardware. Its totally different.
> On the forum, you see that we try to identify problematic versions from sources.
> A clone of the Pm3 v30 Easy [...]

**here is the test result below**

~~~~~~~~~~~~~~~~
██████╗ ███╗   ███╗ ████╗      ...iceman fork          
██╔══██╗████╗ ████║   ══█║       ...dedicated to RDV40           
██████╔╝██╔████╔██║ ████╔╝           
██╔═══╝ ██║╚██╔╝██║   ══█║     iceman@icesql.net          
██║     ██║ ╚═╝ ██║ ████╔╝    https://github.com/rfidresearchgroup/proxmark3/          
╚═╝     ╚═╝     ╚═╝ ╚═══╝  pre-release v4.0          

Support iceman on patreon - https://www.patreon.com/iceman1001/          
                 on paypal - https://www.paypal.me/iceman1001          

          
[=] Session log /home/sa/.proxmark3/log_20191206.txt
[=] Using UART port /dev/ttyACM0           
[=] Communicating with PM3 over USB-CDC           

 [ Proxmark3 RFID instrument ] 
          

 [ CLIENT ]          
  client: RRG/Iceman          
  compiled with GCC 7.4.0 OS:Linux ARCH:x86_64          

 [ PROXMARK3 ]          
          
 [ ARM ]
  bootrom: RRG/Iceman/master/41e0bf9a-dirty-unclean 2019-12-05 02:55:58
       os: RRG/Iceman/master/41e0bf9a-dirty-unclean 2019-12-05 02:56:11
  compiled with GCC 6.3.1 20170620

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev A          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 244188 bytes (47%) Free: 280100 bytes (53%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
~~~~~~~~~~~~~~~~~~~~~

**2)  hw tune command**
~~~~~~~~~~~~~
pm3 --> hw tune
          
[=] Measuring antenna characteristics, please wait...
          
[=] You can cancel this operation by pressing the pm3 button          
..
          
[+] LF antenna: 35.12 V - 125.00 kHz          
[+] LF antenna: 30.87 V - 134.83 kHz          
[+] LF optimal: 36.11 V - 127.66 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 23.33 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 88 is 134.83 kHz, 95 is 125.00 kHz.
~~~~~~~~~~~~~~

**3)    hw status command**
~~~~~~~~~~~~~~~~
[usb] pm3 --> hw status
#db# Memory           
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing           
#db#   tracing ................1          
#db#   traceLen ...............0          
#db# Currently loaded FPGA image           
#db#   mode.................... HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          
#db# LF Sampling config           
#db#   [q] divisor.............95 ( 125.00 kHz )          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db#   [s] samples to skip.....0           
#db# LF T55XX config           
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]          
#db#            mode            |start|write|write|write| read|write|write          
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3          
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------          
#db# fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |           
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |           
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |           
#db#    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |           
#db#           
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......326144          
#db#   Transfer Speed PM3 -> Client = 652288 bytes/s          
#db# Various           
#db#   DBGLEVEL................1          
#db#   ToSendMax...............-1          
#db#   ToSendBit...............0          
#db#   ToSend BUFFERSIZE.......2308          
#db#   Slow clock..............30858 Hz          
#db# Installed StandAlone Mode           
#db#   LF HID26 standalone - aka SamyRun (Samy Kamkar)    
~~~~~~~~~~~~~~~~~~~~~~~~~

4)hf 14a info command

input the command 3 times,with the same card ,no return.(cold)
~~~~~~~~~~~~
[usb] pm3 --> hf 14a info
[usb] pm3 --> hf 14a info
[usb] pm3 --> hf 14a info
~~~~~~~~~~~~~~

(heated )input the command 1 time with the same card,return below 
~~~~~~~~~~~~~~
[usb] pm3 --> hf 14a info
 UID : 84 17 25 02           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1          
[=] proprietary non iso14443-4 card found, RATS not supported          
[+] Magic capabilities : Gen 1a           
[+] Prng detection: WEAK     
~~~~~~~~~~~~~~

iceman1001, 2019-12-06:
That didn't do anything. Its most likely a hw problem. Time for you to follow all lines.  I can't help you.

SDPatrick, 2019-12-06:
> That didn't do anything. Its most likely a hw problem. Time for you to follow all lines. I can't help you.

thanks all the same. time to buy a new one.  Aha...

closed 2019-12-06 by iceman1001