Proxmark/proxmark3 #902: em410x on t55xx chip dump

OscarAkaElvis, 2019-12-25:
Hi, not sure if this could be a bug or maybe I'm doing something wrong. I'm trying to create a dump file from a card but I'm not achieving it. This is the card:

```
proxmark3> lf search
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

EM410x pattern found:

EM TAG ID      : 5D00452C14

Possible de-scramble patterns
Unique TAG ID  : BA00A23428
HoneyWell IdentKey {
DEZ 8          : 04533268
DEZ 10         : 0004533268
DEZ 5.5        : 00069.11284
DEZ 3.5A       : 093.11284
DEZ 3.5B       : 000.11284
DEZ 3.5C       : 069.11284
DEZ 14/IK2     : 00399436491796
DEZ 15/IK3     : 000798874547240
DEZ 20/ZK      : 11100000100203040208
}
Other          : 11284_069_04533268
Pattern Paxton : 1566140948 [0x5D596A14]
Pattern 1      : 8589404 [0x83105C]
Pattern Sebury : 11284 69 4533268  [0x2C14 0x45 0x452C14]

Valid EM410x ID Found!

Valid T55xx Chip Found
Try lf t55xx ... commands
```

I can emulate it and clone to other blank rewritable t55xx chinese card and the cloned card works like a charm. The output of both (original and cloned) is completely the same. Ok but I want to create a dump file from any of them and for some reason I can't get it.

More data:

```
proxmark3> lf t55xx detect
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : No
Block0     : 0x00148040
```

```
proxmark3> lf t55xx info

-- T55x7 Configuration & Tag Information --------------------
-------------------------------------------------------------
 Safer key                 : 0
 reserved                  : 0
 Data bit rate             : 5 - RF/64
 eXtended mode             : No
 Modulation                : 8 - Manchester
 PSK clock frequency       : 0
 AOR - Answer on Request   : No
 OTP - One Time Pad        : No
 Max block                 : 2
 Password mode             : No
 Sequence Start Terminator : No
 Fast Write                : No
 Inverse data              : No
 POR-Delay                 : No
-------------------------------------------------------------
 Raw Data - Page 0
     Block 0  : 0x00148040  00000000000101001000000001000000
-------------------------------------------------------------
```

As you can see, there is no password set. This is what I get when trying to dump it to a file:

```
proxmark3> lf t55xx dump
Reading Page 0:
blk | hex data | binary
----+----------+---------------------------------
  0 | 00148040 | 00000000000101001000000001000000
  1 | FFAB6002 | 11111111101010110110000000000010
  2 | 545C0D24 | 01010100010111000000110100100100
  3 | FFFFFFFF | 11111111111111111111111111111111
  4 | FFFFFFFF | 11111111111111111111111111111111
  5 | FFFFFFFF | 11111111111111111111111111111111
  6 | FFFFFFFF | 11111111111111111111111111111111
  7 | FFFFFFFF | 11111111111111111111111111111111
Reading Page 1:
blk | hex data | binary
----+----------+---------------------------------
  0 | 00148040 | 00000000000101001000000001000000
  1 | E0150A74 | 11100000000101010000101001110100
  2 | 2805A1FC | 00101000000001011010000111111100
  3 | FFFFFFFF | 11111111111111111111111111111111
```

No file is written... I also tried

```
proxmark3> lf t55xx dump f testdump.bin
Reading Page 0:
blk | hex data | binary
----+----------+---------------------------------
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : No
Block0     : 0x00148040

Safety Check: PWD bit is NOT set in config block. Reading without password...
  0 | 00148040 | 00000000000101001000000001000000
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : No
Block0     : 0x00148040

Safety Check: PWD bit is NOT set in config block. Reading without password...
  1 | FFAB6002 | 11111111101010110110000000000010
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : No
Block0     : 0x00148040

Safety Check: PWD bit is NOT set in config block. Reading without password...
  2 | 545C0D24 | 01010100010111000000110100100100
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : No
Block0     : 0x00148040

Safety Check: PWD bit is NOT set in config block. Reading without password...
  3 | FFFFFFFF | 11111111111111111111111111111111
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : No
Block0     : 0x00148040

Safety Check: PWD bit is NOT set in config block. Reading without password...
  4 | FFFFFFFF | 11111111111111111111111111111111
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : No
Block0     : 0x00148040

Safety Check: PWD bit is NOT set in config block. Reading without password...
  5 | FFFFFFFF | 11111111111111111111111111111111
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : No
Block0     : 0x00148040

Safety Check: PWD bit is NOT set in config block. Reading without password...
  6 | FFFFFFFF | 11111111111111111111111111111111
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : No
Block0     : 0x00148040

Safety Check: PWD bit is NOT set in config block. Reading without password...
  7 | FFFFFFFF | 11111111111111111111111111111111
Reading Page 1:
blk | hex data | binary
----+----------+---------------------------------
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : No
Block0     : 0x00148040

Safety Check: PWD bit is NOT set in config block. Reading without password...
  0 | 00148040 | 00000000000101001000000001000000
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : No
Block0     : 0x00148040

Safety Check: PWD bit is NOT set in config block. Reading without password...
  1 | FFAB6002 | 11111111101010110110000000000010
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : No
Block0     : 0x00148040

Safety Check: PWD bit is NOT set in config block. Reading without password...
  2 | 545C0D24 | 01010100010111000000110100100100
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 32
Seq. Term. : No
Block0     : 0x00148040

Safety Check: PWD bit is NOT set in config block. Reading without password...
  3 | FFFFFFFF | 11111111111111111111111111111111
```

I also tried all the commands using `lf t55` instead of `lf t55xx` but got same result.

It is supposed that a file should be written as this thread say: https://github.com/RfidResearchGroup/proxmark3/issues/412

On that thread we can see after a `lf t55 dump` command something like `[+] saved 48 bytes to binary file lf-t55xx-FFCA518B-DEC60C60-data.bin` but is not my case...

What am I'm doing wrong? any help? Thank you.

iceman1001, 2019-12-25:
different repos,  different commands.

OscarAkaElvis, 2019-12-25:
omg... I installed this repo. Has this repo something to dump this kind of cards?

pwpiwi, 2019-12-25:
No. Dumping to file is a new feature which has not yet been merged to this repo. You need to install the RRG repo if you really need it now.

OscarAkaElvis, 2019-12-25:
Thank you for the info. Last question before closing this. Can I install that other repo into another location without interfering to this which is already working?

pwpiwi, 2019-12-26:
The repos won't interfere. But you have to flash the PM3 firmware each time you switch to another repo's client.

OscarAkaElvis, 2019-12-26:
Hmm... that is not an option for me... but, why the command "dump" is appearing for lf t55 and lf t55xx ??? is available. Maybe because is planned to be added in the future? Thanks, closing this.

closed 2019-12-26 by OscarAkaElvis

pwpiwi, 2019-12-26:
The dump command currently dumps to screen only. It shouldn't be very difficult to add the dump to file feature.

OscarAkaElvis, 2019-12-26:
That could be awesome! to create a bin file to store the data and then being able to restore from a bin file to a chameleon or whatever. I know probably you'll have a lot of tasks and features to implement. If you think this is easy could be awesome to be included.

Thanks for your amazing work!