Proxmark/proxmark3 #906: hf search gives error message and wrong tag since update to newest firmware

Lihaes, 2020-01-04:
Hi, 

I have updated the proxmarks boot-loader and fullimage to the newest version:

`Prox/RFID mark3 RFID instrument          
bootrom: master/v3.1.0-170-ga4ff62b-suspect 2020-01-02 18:14:51
os: master/v3.1.0-170-ga4ff62b-suspect 2020-01-02 18:14:41
fpga_lf.bit built for 2s30vq100 on 2019/11/21 at 09:02:37
fpga_hf.bit built for 2s30vq100 on 2019/11/13 at 14:52:19
SmartCard Slot: not available
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 210477 bytes (40). Free: 313811 bytes (60).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
`

Since then, whenever I run a `hf search` I get the following error / tag-found messages, even though there is no card in vicinity:
`proxmark3> hf search
          
#db# Failed to read config block          
#db# Failed to read e-purse          
#db# Failed to read AA block          

14443-3b tag found:          
Unknown tag type answered to a 0x000b3f80 command ans:          
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00           
Abort trap: 6
`

Has somebody else encountered these issues as well and what could I do to resolve this?
I tried to reflash the OS and boot-loader twice, but that didn't help...

Thanks and regards

pwpiwi, 2020-01-05:
Look like you have mixed client and firmware from different repositories?

Lihaes, 2020-01-06:
Hi pwpipw,

actually I freshly pulled the repository, compiled it and installed Boot-Loader and Full-OS according to documentation. I repeated this steps, and have now ga171, but the same problem:

`Prox/RFID mark3 RFID instrument          
bootrom: master/v3.1.0-171-gfef3084-suspect 2020-01-06 21:43:14
os: master/v3.1.0-171-gfef3084-suspect 2020-01-06 21:43:15
fpga_lf.bit built for 2s30vq100 on 2019/11/21 at 09:02:37
fpga_hf.bit built for 2s30vq100 on 2019/11/13 at 14:52:19
SmartCard Slot: not available
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 210477 bytes (40). Free: 313811 bytes (60).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> hf search
          
#db# Failed to read config block          
#db# Failed to read e-purse          
#db# Failed to read AA block          

14443-3b tag found:          
Unknown tag type answered to a 0x000b3f80 command ans:          
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00           
Abort trap: 6`

pwpiwi, 2020-01-07:
Sorry, I misinterpreted the "Abort trap: 6" as an output from the client.

The `#db#` messages are debugging output when looking for an iClass tag. There had been some modifications in the iclass code, therefore still some debugging output.

There is indeed a memory access violation in the iso14443b code. Will fix. Let's see if there is still an answer then...

pwpiwi, 2020-01-07:
Please try PR #908

Lihaes, 2020-01-07:
Hey pwpiwi,

thanks for your fast help! I am not really used to GitHub, so I don't know how to properly get your code changes, but I ve just downloaded the new source-file, added it to the folder and recompiled....
The result is:

`proxmark3> hf search
          
#db# Failed to read config block          
#db# Failed to read e-purse          
#db# Failed to read AA block          
#db# Reading card ...          
No or unknown card found, aborting          

no known/supported 13.56 MHz tags found
          
`

It looks like that solved the issue...

closed 2020-01-07 by pwpiwi

Lihaes, 2020-01-07:
Many thanks again for your support and work!