Proxmark/proxmark3 #907: hf iclass writeblk failure

cjf12harry, 2020-01-06:
I can read block from iclass card, but couldn't write it.

    proxmark> hf iclass readblk b 07 k 0
    CSN: 8d f8 5a 0e fe ff 12 e0          
    Authenticating with legacy diversified key: a8 f8 cd d7 7f 6d ba 50          
    Block 07: c1 37 ee c7 96 d0 79 a3

    proxmark3> hf iclass writeblk b 07 d 1122334455667788 k 0
    CSN: 8d f8 5a 0e fe ff 12 e0          
    Authenticating with legacy diversified key: a8 f8 cd d7 7f 6d ba 50          
    #db# Write block [07] failed          
    Write Block Failed

The following is the trace

    Start |        End | Src | Data (! denotes parity error, ' denotes short bytes)            | CRC | Annotation         |          
    ------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
          0 |       5632 | Rdr | 0a                                                              |     | ACTALL          
      20032 |      22080 | Tag | <SOF>                                                           |     |           
      26176 |      31808 | Rdr | 0c                                                              |     | IDENTIFY          
      35520 |      80576 | Tag | 11  5f  cb  c1  ff  5f  02  bc  b7  e9                          |  ok |           
      84672 |     123072 | Rdr | 81  11  5f  cb  c1  ff  5f  02  bc                              |     | SELECT          
     126720 |     171776 | Tag | 8d  f8  5a  0e  fe  ff  12  e0  6f  1c                          |  ok |           
     830592 |     840320 | Rdr | 88  02                                                          |     | READCHECK[Kd](2)          
     844032 |     880896 | Tag | fe  ff  ff  ff  ff  ff  ff  ff                                  |  ok |           
    1439744 |    1478144 | Rdr | 05  00  00  00  00  d0  e5  f9  0e                              |     | CHECK          
    1481856 |    1502336 | Tag | 53  bc  4c  b3                                                  |  ok |           
    2049728 |    2116800 | Rdr | 87  07  11  22  33  44  55  66  77  88  38  fc  b6  f0  7d  d4  |     | UPDATE(7)          
    2335488 |    2402560 | Rdr | 87  07  11  22  33  44  55  66  77  88  38  fc  b6  f0  7d  d4  |     | UPDATE(7)          
    2621248 |    2688320 | Rdr | 87  07  11  22  33  44  55  66  77  88  38  fc  b6  f0  7d  d4  |     | UPDATE(7)          

Looks like the card didn't response to the update command.

Here is the firware and hardware infor

    proxmark3> hw ver
    Prox/RFID mark3 RFID instrument          
    bootrom: master/v3.1.0-171-gfef3084-suspect 2020-01-04 17:34:12
    os: master/v3.1.0-171-gfef3084-suspect 2020-01-04 17:34:14
    fpga_lf.bit built for 2s30vq100 on 2019/11/21 at 09:02:37
    fpga_hf.bit built for 2s30vq100 on 2019/11/13 at 14:52:19
    SmartCard Slot: not available
          
    uC: AT91SAM7S512 Rev B          
    Embedded Processor: ARM7TDMI          
    Nonvolatile Program Memory Size: 512K bytes. Used: 209347 bytes (40%). Free: 314941 bytes(60%).          
    Second Nonvolatile Program Memory Size: None          
    Internal SRAM Size: 64K bytes          
    Architecture Identifier: AT91SAM7Sxx Series          
    Nonvolatile Program Memory Type: Embedded Flash Memory

pwpiwi, 2020-01-06:
I cannot yet confirm this issue. Writing blocks works fine for me. Can you please send the output from `hf iclass reader 1` ? What type of iClass card do you use?

cjf12harry, 2020-01-07:
CSN: xx xx xx 0c fe ff 12 e0
     CC: fe ff ff ff ff ff ff ff
     Mode: Application [Locked]
     Coding: ISO 14443-2 B/ISO 15693
     Crypt: Secured page, keys not locked
     RA: Read access not enabled
     Mem: 2 KBits/2 App Areas (32 * 8 bytes) [1F]
     AA1: blocks 06-12
     AA2: blocks 13-1F
     AppIA: ff ff ff ff ff ff ff ff
      : Possible iClass (legacy tag)

pwpiwi, 2020-01-07:
I cannot see anything special with your tag's config. Both authentication and reading of block 7 are allowed. Same key used to authenticate read and write.

Your antenna is OK - your trace doesn't show any communication issue.

At the moment I can only think of that your card responds exceptionally slow to write commands. Can you please try to increase `ICLASS_READER_TIMEOUT_UPDATE` in armsrc/iclass.c? You can see if your modification was successful, when the time between the three Update commands in your trace increases accordingly (or if the card responds).

cjf12harry, 2020-01-08:
Start |        End | Src | Data (! denotes parity error, ' denotes short bytes)            | CRC | Annotation         |          
    ------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
          0 |       5632 | Rdr | 0a                                                              |     | ACTALL          
       9344 |      11392 | Tag | <SOF>                                                           |     |           
      15488 |      21120 | Rdr | 0c                                                              |     | IDENTIFY          
      24832 |      69888 | Tag | xx  xx  xx  c1  ff  5f  02  1c  b7  4d                          |  ok |           
      73984 |     112384 | Rdr | xx  xx  xx  89  c1  ff  5f  02  1c                              |     | SELECT          
     116096 |     161152 | Tag | xx  xx  xx  0c  fe  ff  12  e0  5e  83                          |  ok |           
     789568 |     799296 | Rdr | 88  02                                                          |     | READCHECK[Kd](2)          
     803008 |     839872 | Tag | fe  ff  ff  ff  ff  ff  ff  ff                                  |  ok |           
    1398848 |    1437248 | Rdr | 05  00  00  00  00  d9  0a  52  84                              |     | CHECK          
    1440960 |    1461440 | Tag | 97  9a  d7  78                                                  |  ok |           
    2008704 |    2075776 | Rdr | 87  07  01  02  03  04  05  06  07  08  62  3a  e2  11  18  f9  |     | UPDATE(7)          
    3526464 |    3593536 | Rdr | 87  07  01  02  03  04  05  06  07  08  62  3a  e2  11  18  f9  |     | UPDATE(7)          
    5044224 |    5111296 | Rdr | 87  07  01  02  03  04  05  06  07  08  62  3a  e2  11  18  f9  |     | UPDATE(7)          

Above is the result of extending the retry delay. The time spacing grows from 16ms to 108ms. Still there is no response from the card.

pwpiwi, 2020-01-08:
In order to rule out antenna issues I did some checks with a Proxmark RDV3 Easy and measured the maximum possible antenna distance for both readblk and writeblk operations. I could do reliable reads and writes at a maximum distance of 6.3 mm. We therefore can even exclude that write operations draw substantially more power than read operations and therefore could cause the issues.

This is at least true for my iClass DP * and iClass DL * cards. Which type of card do you have under test?

cjf12harry, 2020-01-08:
I am using iClass tags, those blue key fob. I don't have more information about these keys than the read info blocks I post above. Do you know how to get lower level debug information than the log trace? I want to see if the tag responded (but not received by proxmark client).

pwpiwi, 2020-01-09:
Fobs are usually more tricky than cards because of their smaller antenna size. But coupling seems to be OK - your trace looks good and the communication for writing isn't different from reading (reader sends a command and tag answers in both cases).

For lower level debugging I can think of two options:
* you can try to pick up the communication with a second proxmark using 'hf snoop'. It might be hard to find the correct parameters (triggers to skip, samples to skip).
* the proxmark maintains a ring buffer capturing the raw signal within the last 1.8ms of each communication. It can be displayed with `hf plot`. With the default setting of `ICLASS_READER_TIMEOUT_UPDATE` this would be the signal from approx. 14.2 ms to 16ms after the last Update command has been sent, i.e. you would most probably see nothing if the tag had already responded earlier. By stepwise reducing `ICLASS_READER_TIMEOUT_UPDATE` you could piece together the response signal (if there is any). Use `hf plot` after a successful `hf iclass readblk` to see how a signal could look like and how strong it could be.

x225am, 2020-01-09:
I have exactly same issue.

steve-embling, 2020-01-10:
Hey, I'm wondering if this is relating to issue #820 ?
Do you have a card listed in that page - I still haven't been able to test writing since that fix..

cjf12harry, 2020-01-11:
Currently I don't have the iclass 2000, need to order some to continue testing.

steve-embling, 2020-01-11:
There's a description in there of the PicoPass chip being discontinued so the breaking change might be trickling into other lines too.

It would be possible to test this without ordering new cards by reverting to that older version and seeing if that old version breaks read capability too. Having said that - I might be completely wasting your time here. So I'm sorry in advance if that is the case.

pwpiwi, 2020-01-12:
@steve-embling: 
1. I don't think that this is related to #820. Reading seems to work for all kind of cards. At least I didn't see any negative reports.
2. I might be naive, but I cannot imagine that HID changes their credentials such that they cannot be used any longer by all the existing HID readers in the field. Whatever chip they are using now, it should be backwards compatible with the PicoPass chip.

steve-embling, 2020-01-12:
Fair enough. For point 2 it does seem conceivable there could be subtle accidental changes in timing or behaviour that would break some kind of assumption made by the Proxmark but still maintain backwards compatibility with official readers.

The PR that appeared to fix the referenced issue was quite busy so I've no idea what the change was that got the card I could previously not read to suddenly start reading (it wasn't my own card so I couldn't test anything potentially destructive like writing). But the lack of response on the other post led me to believe that maybe it wasn't a common issue and writing hadn't been tested yet (and potentially why it hasn't been closed). I just stumbled across this issue after checking on the status of the other.

But yes, happy to accept I'm completely wrong here, and apologies.

pwpiwi, 2020-01-12:
I couldn't find one of those problematic cards, i.e. up to now I could read all iClass cards even before the respective PRs, although not very reliable. And I didn't receive a conclusive feedback from anyone that issue #820 is fixed. I therefore keep it open until I find a card which shows the issue or someone confirms that a card couldn't be read before and can be read after the fix. If you can confirm that, I will close #820.

Regarding point 2 I would agree with your assumption. The Proxmark showed some weird behaviour with the iClass implementation and I believe that most iClass operations succeeded only by chance and because of many retries. I cannot really prove this statement because at this time there had not been a usable iClass tracing as well.

There is also the possibility that those problematic cards have ISO15693 disabled and communicate using ISO14443B only. Proxmark doesn't support those iClass cards yet.

vpr1982, 2020-03-04:
Same issue here. Unable to write.