Proxmark/proxmark3 #914: Waiting from proxmark to appear

RJswap, 2020-02-17:
Hi !

I am a new user of proxmark3. I therefore followed every steps of the wiki tutorial to flash my proxmark. I tried the procedure on both kali and ubuntu but when I flash I got the message"waiting from proxmark to appear on /dev/ttyACM0...". (The dmesg command worked as expected and the proxmark is seen on ttyACM0). First I was running on a windows subsystem but then i tried on a linux (ubuntu) laptop but the problem remains. I tried some solutions found on internet such as using sudo but none of them solved the issue.
Do you know what could be wrong with my attempt ?

Thanks for your help

pwpiwi, 2020-02-17:
Did you read about modem_manager?

RJswap, 2020-02-17:
Thanks for your reply,
Yes I tried the solution suggested on proxmark.org about modem_manager

pwpiwi, 2020-04-07:
Could you solve your problem? If yes, would you mind to tell the community what went wrong in first place?

RJswap, 2020-04-07:
Hi,
No, I still could not solve my problem. 
at first I used an ubuntu virtual machine on windows (I thought there were problems with authorization of access to usb devices); then i installed a whole ubuntu os on a new machine but the problem remains the same

pwpiwi, 2020-04-07:
Please send us some screenshots or console cut&paste from dmesg, proxmark3, and flasher commands. Both your input and the respective output.

pwpiwi, 2020-05-06:
Are you still there?

Aquarh, 2020-05-06:
Yes but with confinement I no longer have access to my proxmark so I am not able to provide screenshots. I haven't given up x)

seanjwaters, 2021-03-23:
I'm currently having the same issues. I'm following [this guide](https://hackerwarehouse.tv/product-knowledgebase/proxmark/proxmark3-rdv4-setup-in-kali-linux/). Everything seems to be normal until I get to the _Flashing the Proxmark3 RDV4_ section. Screenshots linked below.

[dmesg](https://i.imgur.com/Cfh0thF.png)
[image]
[flashing-error](https://i.imgur.com/6YvP5kJ.png)

iceman1001, 2021-03-23:
I suggest following the instructions on RRG/Iceman repository if you want to install it on your RDV4,  for the most up-to-date guides.