Proxmark/proxmark3 #915: Compiling in Ubuntu - Comms.c getCommand error

orlando2k7, 2020-02-22:
I am new to writing issues on Github so please bear with me. I was trying to compile the client in ubuntu 16.04 using the latest gcc 5 and a previous version gcc 4.7 with no luck. 

Using the make clean && make all command to run through the building process and it errors out when it gets to the 'comms.c' file, specifically on the 'getCommand' function.

Error that shows up just before it exits with a recipe for target failed:

comms.c: In function ‘getCommand’:
comms.c:146:18: error: storage size of ‘end_time’ isn’t known
comms.c:147:2: warning: implicit declaration of function ‘clock_gettime’ [-Wimplicit-function-declaration]
comms.c:147:16: error: ‘CLOCK_REALTIME’ undeclared (first use in this function)
comms.c:147:16: note: each undeclared identifier is reported only once for each function it appears in
comms.c:146:18: warning: unused variable ‘end_time’ [-Wunused-variable]
Makefile:348: recipe for target 'obj/comms.o' failed
make[1]: *** [obj/comms.o] Error 1
make[1]: Leaving directory '/home/prox/proxmark3/client'
Makefile:33: recipe for target 'client/all' failed
make: *** [client/all] Error 2

I am not quite sure what exactly is causing this issue or how to fix it. Is anyone else having this problem trying to compile the client?

pwpiwi, 2020-02-24:
Could you please try if PR #917 fixes your issue.

alvra, 2020-03-06:
Same issue here. With #917 that still leaves the error of `end_time` at line 147 (just before the changes in the PR).

```
comms.c: In function ‘getCommand’:
comms.c:147:18: error: storage size of ‘end_time’ isn’t known
  struct timespec end_time;
                  ^~~~~~~~
comms.c:147:18: warning: unused variable ‘end_time’ [-Wunused-variable]
```

pwpiwi, 2020-03-06:
pushed. Maybe it is time to move to C11 instead of 20 years old C99.

pwpiwi, 2020-03-17:
The fix (PR #917) passes Travis CI. Can someone confirm that it works in real life as well?

pwpiwi, 2020-04-07:
Taking no feedback as positive feedback. Merged #917.

closed 2020-04-07 by pwpiwi