Proxmark/proxmark3 #916: ISO15693 reading/emulation stability

g3gg0, 2020-02-22:
Hello dev team.

Recently I added SLIX-L features (emulation and reading) to a custom fork of the proxmark project.
Adding the features was quite easy and everything went fine until I wanted to rebase the changes to a more recent commit. (for maybe starting a PR)
Suddenly the emulation didn't work at all, looked like there are a lot of decoding errors or whatever.
Also reading SLIX-L tags didnt work properly anymore - had to perfectly align the tag to get a glimpse of a chance to read them.
Going back to the "old" commit of the proxmark repository solved the issues again and everything worked like a charm. Could move the tag over centimeters without any impact on read results.
With the "bad" commits, playing with the tag showed that I indeed *can* read the tag, if it is perfectly aligned. 
Just moving it a few mm causes a lot of readout errors or no detection at all.
This was a perfect setup for a bisect session.

So I set up the proxmark with a ISO15693 tag on top and built a script which compiles and flashes the currently checked out version to the proxmark and does 50 test reads of the tag.
After running the bisect, I ended up with a sharp transition of "does not read this tag in this position" and "does the read quite well".

[image]

bad: c41dd5f9f690777eefd08eaeb43db52f599ce38c
good: b41be3cb11103739d2fa92ced09d1e29f8151b7b

**Question:**
Can someone from dev team with an ISO15693 tag confirm that there is some trouble with reading them reliable? (use "good" and place the tag that is just barely can read it and then check "bad")

**Setup:**
Windows 10 (WSL: Ubuntu bionic)
Hardware: proxmark3 easy (chinese clone probably)
HF tune: 30699 mV
Tag: ISO15693, SLIX-L

pwpiwi, 2020-02-23:
I stumbled over another error the day before yesterday. iClass simulation doesn't work anymore and I found the same commit introducing this error (I did it manually, your script would have been very handy 😃). Please note that c41dd5f ("bad") is older than b41be3c ("good"). It has been merged later with the merge commit e55b441. So it might be a merge error.

g3gg0, 2020-02-23:
ah okay nice :)

here a screenshoot of the log. bad is indeed older. didn't expect that.

[image]

g3gg0, 2020-02-24:
Do you need any further information to work on this issue?
Would it help to provide some access to my proxmark w/ NFC tag so you could use it to reproduce the issue?
(e.g. TCP port -> character device forwarding)

regards,
g3gg0

pwpiwi, 2020-02-26:
I have all I need to work on the issue except some time. If you have a hint please post it here.

g3gg0, 2020-02-28:
ok some result:
reverting the Handle15693SamplesFromTag() to the previous version from b41be3c resolves the issue.

g3gg0, 2020-03-29:
i guess you didnt have time to check this?

pwpiwi, 2020-03-29:
I have prepared some fixes a few weeks ago but then got distracted by vacation and covid-19. Would you want to test?

g3gg0, 2020-03-29:
sure, will test it of course.

g3gg0, 2020-03-31:
Hi, thanks for working on it.

setup: 
 - placed the tag on the reader
 - tried with two commits without moving anything

result:
```
# git checkout -f b41be3c && ../test.sh
Checking out files: 100% (1173/1173), done.
HEAD is now at b41be3c Merge pull request #862 from pwpiwi/fix_iclass_sim
 WORKING (26/50 clean reads)

# git checkout -f f66d28af && ../test.sh
Checking out files: 100% (1173/1173), done.
Previous HEAD position was b41be3c Merge pull request #862 from pwpiwi/fix_iclass_sim
HEAD is now at f66d28a Merge branch 'master' into fix_iso15693_fpga
 FAILED  (0/50 clean reads)
```

pwpiwi, 2020-04-01:
Up to now I had overlooked that you never had reliable readouts (your best was 42/50 for a "good" commit - which I would consider quite unreliable).

I thought I should do some tests using your script but made the following modifications to it:
- removed the output redirections. I want to see errors and warnings.
- removed the `make -j`. The Makefiles have never been checked for parallel processing.

You should not "place the tag on the reader" but keep some distance. And if you say "tag" what exactly does this mean? If it is not a card you better should use some smaller antenna (which is not an option with RDV3 Easy). I used a testing setup with a card approx 2cm above the antenna:

[image]

I got 50/50 clean reads with both b41be3c and f66d28af on all Orignal Proxmark, a Proxmark RDV2, and a Proxmark RDV4.
 
I got 3/50 or even 2/50 clean reads with b41be3c but 50/50 clean reads with f66d28af on a Proxmark RDV3 Easy.

So it seems to be another of these infamous RDV3 Easy issues.

g3gg0, 2020-04-02:
Fine you can reproduce it on your desk.
So what if you place RDV4 + f66d28a so that you wont get 50/50 and then test b41be3c ?
If this gives the same results, then it is indeed a weird rdv3 easy issue.

My setup is an easy with a figurine on it, that has a tag inside.

iceman1001, 2020-04-03:
There a bit confusing going on here. 
Offical repo has @pwpiwi  's changes to fpga/arm/client for demodulate HF signals.
RRG/Iceman repo doesn't have those yet.    Mixing fixes for offical repo just like that into RRG/Iceman is not useful.

g3gg0, 2020-04-03:
hello iceman.
i am a bit confused - which changes should go into iceman repo?

iceman1001, 2020-04-03:
None, 
just to make sure I are on the same page,
pull latest from offical repo,
compile, flash,
--
Now you can test all code from piwi.
..
If you already had that setup,  then it was my missunderstanding you.

g3gg0, 2020-04-03:
ah okay. no problem.
i was working on Proxmark/proxmark3 and made my comparisons on commits in this repository.
pwipwi brought a fix into the game which was on pwpiwi/proxmark3 (branch fix_iso15693_fpga)
this fix didnt change the read stability with my setup.

I did not try your repository yet, but it looked like you didn't have the patches which caused the read stability to drop.
so your repository would most likely give the same results as b41be3c on the official repo.
i guess :)

pwpiwi, 2020-04-03:
> Fine you can reproduce it on your desk.

No, I can't. With RDV3 Easy I have reliable readings with latest commit (f66d28a) and very unreliable readings with b41be3c. It is the other way round for you. And it is the other way round for me with RDV4 at greater distances (read on).

> So what if you place RDV4 + f66d28a so that you wont get 50/50 (maybe 25/50) and then test b41be3c ?

f66d28a on RDV4 starts to fail at a distance of approx. 7.5cm (32/50 clean reads). b41be3c gives 50/50 at the same distance. RDV3 fails completely at this distance (0/50 regardless of commit). 

Hmm.

pwpiwi, 2020-04-03:
@iceman1001:
>Mixing fixes for offical repo just like that into RRG/Iceman is not useful.

With "RDV4 + f66d28a" @g3gg0 meant f66d28a on RDV4 hardware, not on RRG repo (at least this is what I understood).

g3gg0, 2020-04-03:
> [f66d28a](https://github.com/Proxmark/proxmark3/commit/f66d28afcc699588612e40359e02d75eb408739c) on RDV4 starts to fail at a distance of approx. 7.5cm (32/50 clean reads). [...]

so can we conclude:
-> the RDV3 easy has a worse antenna (tuning) compared to RDV4 ?

the tags i am reading, i cannot position "perfectly" so that i get the best possible reading.

and as i compared the b41be3c to your fix - but not your fix vs the worst, i cannot say if it got "better".
it is just not as good as the old b41be3c (yet?)

i will try all three variants

pwpiwi, 2020-04-03:
please wait a few minutes...

pwpiwi, 2020-04-03:
> so can we conclude:
>-> the RDV3 easy has a worse antenna (tuning) compared to RDV4 ?

The forum and github issues are full with RDV3 Easy issues especially when sniffing. Could be the antenna or antenna tuning or some bad circuit design or low quality components or whatever...

I have pushed a new commit (6792435) to PR #926. With this I am getting the following results:

RDV4 with 6792435: starts failing at 10cm (30/50). 0/50 with b41be3c at this distance.
RDV3 Easy with 6792435: starts failing at 7.5cm (25/50). 0/50 with b41be3c at this distance.

g3gg0, 2020-04-04:
jackpot!
now reading is a lot more stable and seems to work.
will do a few more tests :)

```
> for commit in b41be3c f66d28a 6792435; do echo "testing $commit..."; echo "----------------------"; git checkout -f $commit > /dev/null 2>&1; ../test.sh; echo ""; done
testing b41be3c...
----------------------
 FAILED  (13/50 clean reads)

testing f66d28a...
----------------------
 FAILED  (0/50 clean reads)

testing 6792435...
----------------------
 WORKING (50/50 clean reads)
```

closed 2020-04-11 by g3gg0