Proxmark/proxmark3 #918: #678 breaks [hw tune] high frequency voltage readings on some platforms

0x2b3bfa0, 2020-02-27:
Since the merge of #678, the high frequency antenna voltage readings from `hw tune` are far below the values reported by older firmware versions. The device used for testing purposes was _Elechouse Proxmark3 Easy_, but this might be happening on all the non-RDV4 devices.

***

**`hw tune` high frequency readings _before_ 050aa18b13084b0d57ffcee352bf7a30b828b8f5**

```
# LF antenna: 26.12 V @   125.00 kHz
# LF antenna: 31.49 V @   134.00 kHz
# LF optimal: 33.55 V @   130.43 kHz
# HF antenna: 36.26 V @    13.56 MHz
```

**`hw tune` high frequency readings _after_ 050aa18b13084b0d57ffcee352bf7a30b828b8f5**

```
# LF antenna: 26.26 V @   125.00 kHz
# LF antenna: 31.62 V @   134.00 kHz
# LF optimal: 33.55 V @   130.43 kHz
# HF antenna:  6.24 V @    13.56 MHz
# Your HF antenna is marginal.
```

pwpiwi, 2020-03-04:
Did you rebuild the client after flashing the firmware?

0x2b3bfa0, 2020-03-10:
Yes, as far as `make clean` is able to delete the old binaries.

pwpiwi, 2020-03-16:
Sorry, but I cannot confirm the issue. I compared the most current HEAD (5bc3841) with the release before 050aa18 (which is 2758d83) on three different PM3s (Original, RDV2, RDV3 Easy). HF voltage results look OK and consistent:
```
PM3          |   5bc3841    |   2758d83 
-------------+--------------------------
Original     |    22.05V    |    22.12V
RDV2         |    32.26V    |    32.47V
RDV3 Easy    |    30.91V    |    30.91V
```

0x2b3bfa0, 2020-03-16:
I've bought my PM3 Easy (blue circuit board) on AliExpress, and chances are that it was a knockoff. I've sent it back to the seller for a full refund, just in case it was a hardware issue, and I'll repeat the tests after getting a new device.

What a strange issue!

closed 2020-04-07 by pwpiwi