Proxmark/proxmark3 #921: Error occurs compiling comms.c in kali

cqqjlxy, 2020-03-16:
I had an error when I was compiling comms.c in kali
comms.c: In function ‘getCommand’:
comms.c:147:2: warning: implicit declaration of function ‘clock_gettime’ [-Wimplicit-function-declaration]
  147 |  clock_gettime(CLOCK_REALTIME, &end_time);
      |  ^~~~~~~~~~~~~
comms.c:147:16: error: ‘CLOCK_REALTIME’ undeclared (first use in this function)
  147 |  clock_gettime(CLOCK_REALTIME, &end_time);
      |                ^~~~~~~~~~~~~~
comms.c:147:16: note: each undeclared identifier is reported only once for each function it appears in
Anyone knows how to solve it??

iceman1001, 2020-03-16:
Are you running on latest Kali?   Which is your gcc --version?

And offical repo has a similar thread.  https://github.com/Proxmark/proxmark3/issues/915

pwpiwi, 2020-03-16:
Duplicate of issue #915. Did you try the fix in PR #917 ?

closed 2020-03-16 by pwpiwi