Proxmark/proxmark3 #922: /System/Library/Extensions/Proxmark3.kext/Contents: Read-only file system

echo-cool, 2020-03-19:
While I try to use `sudo make install_kext`
There is an error.

MacOS 10.15.3

```
mkdir -p /System/Library/Extensions/Proxmark3.kext/Contents
mkdir: /System/Library/Extensions/Proxmark3.kext/Contents: Read-only file system
make: *** [Makefile:39: install_kext] Error 1
```

[image]

pwpiwi, 2020-03-19:
You are walking on ancient grounds :smiley:. I don't own or know much about MacOS. If this compares to Linux then probably the respective filesystem needs to be remounted read/write before `make install_kext`. However, I would assume that there is a more standard way to add Extensions...

pwpiwi, 2020-03-19:
Please try Pull Request #923 which should do the remount when running 'sudo make kext_install'

iceman1001, 2020-03-19:
...like Piwi said,   why are you using the **hid-flasher** ?  We are talking pre-2013 now.

echo-cool, 2020-03-19:
while, the firmware of my proxmark3 is quite old.... I just follow the instructions to upgrade....
[image]

> Please try Pull Request #923 which should do the remount when running 'sudo make kext_install'
Thanks, Let me try.

echo-cool, 2020-03-19:
> ...like Piwi said, why are you using the **hid-flasher** ? We are talking pre-2013 now.

It seems like it still can't work. Maybe to make `/system` unchangeable is the new feature of Catalina to protect the MacOS ?
[image]

echo-cool, 2020-03-19:
oh, I solved this problem by switching system.
Thanks

pwpiwi, 2020-03-21:
Switching the system is not really a fix. 😃 

I have added the root directory to the mount command. If you don't mind, can you please `sudo make install_kext` again. Some users may benefit in the future.

alfonsrv, 2020-06-15:
Still does not work, same error. Upgrade wiki for macOS seems outdated.

pwpiwi, 2020-06-15:
You are obviously using the RRG repository and therefore should report issues there: https://github.com/RfidResearchGroup/proxmark3/issues

alfonsrv, 2020-06-15:
No. I tried using the RRG repository before and – after I got the error my board would not support the commands yet, I thought I would have to update the bootloader manually before I could use the more "automatic" update process.  
So I resorted to the Proxmark/proxmark3 repo and followed the macOS steps, which puts me in the position to confirm the kext issue is still very much present under `macOS 10.15.5`. Ignoring the HID flasher and the connected kext, instead using the `client/flasher` worked though.

Afterwards I ran into an issue where the `proxmark3` cli client would no longer talk to my Proxmark, before I read somewhere that the clients are not intercompatible with the firmware anymore. So this issue was ultimately resolved by using the RRG repository instead and manually flashing their image instead.  
Besides the discrepancies with macOS Catalina clearly not being supported properly in the process outlined in the wiki, I must say it is somewhat confusing to have no note anywhere how much RRG and this repo have formed a defacto two-world system, other than on some nested sub-pages and forum posts.

iceman1001, 2020-06-15:
....
if you get that message.... 
you should flash bootrom first like the instructions says.

```
./pm3-flash-bootrom
./pm3-flash-fullimage
```

Its the _normal_ way to switch to RRG/Iceman repo.

pwpiwi, 2020-06-15:
> which puts me in the position to confirm the kext issue is still very much present under macOS 10.15.5

Did you apply the changes in https://github.com/Proxmark/proxmark3/pull/923/files before running `sudo make install_kext`?

alfonsrv, 2020-06-15:
> Did you apply the changes in https://github.com/Proxmark/proxmark3/pull/923/files before running `sudo make install_kext`?

Sorry, thought it was already merged. I did not.

pwpiwi, 2020-06-15:
No it isn't merged. I don't have access to a MacOS machine and I don't like to merge untested changes.

alfonsrv, 2020-06-15:
Probably a setup Virtualbox image out there. But I'll try the code on a lab Hackintosh on Wednesday and will report back.

mattop4d, 2020-09-17:
Came here to confirm that the fix in the PR is not working. I got this error:

```
$ sudo make install_kext

spctl --master-disable
mount -uw /
mount_apfs: volume could not be mounted: Operation not permitted
mount: / failed with 77
make: *** [install_kext] Error 77
```

Tested on a Macbook Pro with MacOS Catalina 10.15.6

pwpiwi, 2020-10-20:
See https://www.tonymacx86.com/threads/macos-10-15-catalina-how-to-mount-root-partition-read-write-for-post-installation.283749/page-2#post-2018467 for some hint