Proxmark/proxmark3 #924: proxmark on ubuntu 18.04 64bit error by CLOCK_REALTIME

linux0ne-su, 2020-03-27:
(base) ooo0000oo@Guo-desktop:~/proxmark3$ make
make -C client  all
make[1]: 进入目录“/home/oooo0000o/proxmark3/client”
Package libpcsclite was not found in the pkg-config search path.
Perhaps you should add the directory containing `libpcsclite.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libpcsclite' found
Package libpcsclite was not found in the pkg-config search path.
Perhaps you should add the directory containing `libpcsclite.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libpcsclite' found
Compiling liblua, using platform linux
cd ../liblua && make linux
make[2]: 进入目录“/home/oo0000ooo/proxmark3/liblua”
make all SYSCFLAGS="-DLUA_USE_LINUX" SYSLIBS="-Wl,-E -ldl -lreadline -ltermcap -lncurses"
make[3]: 进入目录“/home/ooo0000oo/proxmark3/liblua”
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lapi.o lapi.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lcode.o lcode.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lctype.o lctype.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o ldebug.o ldebug.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o ldo.o ldo.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o ldump.o ldump.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lfunc.o lfunc.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lgc.o lgc.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o llex.o llex.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lmem.o lmem.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lobject.o lobject.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lopcodes.o lopcodes.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lparser.o lparser.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lstate.o lstate.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lstring.o lstring.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o ltable.o ltable.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o ltm.o ltm.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lundump.o lundump.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lvm.o lvm.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lzio.o lzio.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lauxlib.o lauxlib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lbaselib.o lbaselib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lbitlib.o lbitlib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lcorolib.o lcorolib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o ldblib.o ldblib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o liolib.o liolib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lmathlib.o lmathlib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o loslib.o loslib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lstrlib.o lstrlib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o ltablib.o ltablib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o loadlib.o loadlib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o linit.o linit.c
ar rcu liblua.a lapi.o lcode.o lctype.o ldebug.o ldo.o ldump.o lfunc.o lgc.o llex.o lmem.o lobject.o lopcodes.o lparser.o lstate.o lstring.o ltable.o ltm.o lundump.o lvm.o lzio.o lauxlib.o lbaselib.o lbitlib.o lcorolib.o ldblib.o liolib.o lmathlib.o loslib.o lstrlib.o ltablib.o loadlib.o linit.o 
ar: `u' 修饰符被忽略，因为 `D' 为默认（参见 `U'）
ranlib liblua.a
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lua.o lua.c
gcc -o lua   lua.o liblua.a -lm -Wl,-E -ldl -lreadline -ltermcap -lncurses 
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o luac.o luac.c
gcc -o luac   luac.o liblua.a -lm -Wl,-E -ldl -lreadline -ltermcap -lncurses 
make[3]: 离开目录“/home/ooooo/proxmark3/liblua”
make[2]: 离开目录“/home/ooooo/proxmark3/liblua”
Compiling jansson
cd ./jansson && make all
make[2]: 进入目录“/home/ooooo/proxmark3/client/jansson”
gcc -DHAVE_STDINT_H -I. -O2 -Wall -Wno-unused-variable -Wno-unused-function -c -o dump.o dump.c -lm  
gcc -DHAVE_STDINT_H -I. -O2 -Wall -Wno-unused-variable -Wno-unused-function -c -o error.o error.c -lm  
gcc -DHAVE_STDINT_H -I. -O2 -Wall -Wno-unused-variable -Wno-unused-function -c -o hashtable.o hashtable.c -lm  
gcc -DHAVE_STDINT_H -I. -O2 -Wall -Wno-unused-variable -Wno-unused-function -c -o hashtable_seed.o hashtable_seed.c -lm  
gcc -DHAVE_STDINT_H -I. -O2 -Wall -Wno-unused-variable -Wno-unused-function -c -o load.o load.c -lm  
gcc -DHAVE_STDINT_H -I. -O2 -Wall -Wno-unused-variable -Wno-unused-function -c -o memory.o memory.c -lm  
gcc -DHAVE_STDINT_H -I. -O2 -Wall -Wno-unused-variable -Wno-unused-function -c -o pack_unpack.o pack_unpack.c -lm  
gcc -DHAVE_STDINT_H -I. -O2 -Wall -Wno-unused-variable -Wno-unused-function -c -o strbuffer.o strbuffer.c -lm  
gcc -DHAVE_STDINT_H -I. -O2 -Wall -Wno-unused-variable -Wno-unused-function -c -o strconv.o strconv.c -lm  
gcc -DHAVE_STDINT_H -I. -O2 -Wall -Wno-unused-variable -Wno-unused-function -c -o utf.o utf.c -lm  
[... 71 lines cut ...]
mv -f obj/uart_posix.Td obj/uart_posix.d
gcc -MT obj/uart_win32.o -MMD -MP -MF obj/uart_win32.Td  -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I/opt/local/include -I../zlib -I../uart -I../liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -g -O3  -DWITH_ISO14443a_StandAlone -DWITH_LF -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG -DWITH_CRC -DWITH_HFSNOOP -DWITH_SMARTCARD -DWITH_GUI -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED   -c -o obj/uart_win32.o ../uart/uart_win32.c
mv -f obj/uart_win32.Td obj/uart_win32.d
gcc -MT obj/util.o -MMD -MP -MF obj/util.Td  -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I/opt/local/include -I../zlib -I../uart -I../liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -g -O3  -DWITH_ISO14443a_StandAlone -DWITH_LF -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG -DWITH_CRC -DWITH_HFSNOOP -DWITH_SMARTCARD -DWITH_GUI -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED   -c -o obj/util.o util.c
mv -f obj/util.Td obj/util.d
gcc -MT obj/util_posix.o -MMD -MP -MF obj/util_posix.Td  -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I/opt/local/include -I../zlib -I../uart -I../liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -g -O3  -DWITH_ISO14443a_StandAlone -DWITH_LF -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG -DWITH_CRC -DWITH_HFSNOOP -DWITH_SMARTCARD -DWITH_GUI -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED   -c -o obj/util_posix.o util_posix.c
mv -f obj/util_posix.Td obj/util_posix.d
gcc -MT obj/ui.o -MMD -MP -MF obj/ui.Td  -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I/opt/local/include -I../zlib -I../uart -I../liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -g -O3  -DWITH_ISO14443a_StandAlone -DWITH_LF -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG -DWITH_CRC -DWITH_HFSNOOP -DWITH_SMARTCARD -DWITH_GUI -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED   -c -o obj/ui.o ui.c
mv -f obj/ui.Td obj/ui.d
gcc -MT obj/comms.o -MMD -MP -MF obj/comms.Td  -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I/opt/local/include -I../zlib -I../uart -I../liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -g -O3  -DWITH_ISO14443a_StandAlone -DWITH_LF -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG -DWITH_CRC -DWITH_HFSNOOP -DWITH_SMARTCARD -DWITH_GUI -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED   -c -o obj/comms.o comms.c
comms.c: In function ‘getCommand’:
comms.c:147:2: warning: implicit declaration of function ‘clock_gettime’; did you mean ‘localtime’? [-Wimplicit-function-declaration]
  clock_gettime(CLOCK_REALTIME, &end_time);
  ^~~~~~~~~~~~~
  localtime
comms.c:147:16: error: ‘CLOCK_REALTIME’ undeclared (first use in this function)
  clock_gettime(CLOCK_REALTIME, &end_time);
                ^~~~~~~~~~~~~~
comms.c:147:16: note: each undeclared identifier is reported only once for each function it appears in
Makefile:348: recipe for target 'obj/comms.o' failed
make[1]: *** [obj/comms.o] Error 1
make[1]: 离开目录“/home/ooooo/proxmark3/client”
Makefile:33: recipe for target 'client/all' failed
make: *** [client/all] Error 2
(base) ooo0000oo@Guo-desktop:~/proxmark3$

linux0ne-su, 2020-03-28:
是因为变量tv没有声明么？

pwpiwi, 2020-03-28:
Did you try the fix in PR #917 ?

linux0ne-su, 2020-03-30:
@pwpiwi 
Yes.  This is how I operate.
Detailed steps ：
Add the following code to the commas.c file, save and compile.

-----------------------------code----------------------------------------------------
#define _POSIX_C_SOURCE 199309L // need clock_gettime()

#include "comms.h"

#include <stdio.h>

static int getCommand(UsbCommand response, uint32_t ms_timeout) {

struct timespec end_time;
#if (_POSIX_TIMERS > 0)
clock_gettime(CLOCK_REALTIME, &end_time);
#else
struct timeval tv;
gettimeofday(&tv, NULL);
end_time.tv_sec = tv.tv_sec;
end_time.tv_nsec = tv.tv_usec * 1000;
#endif
end_time.tv_sec += ms_timeout / 1000;
end_time.tv_nsec += (ms_timeout % 1000) * 1000000;
if (end_time.tv_nsec > 1000000000) {
...........

pwpiwi, 2020-03-30:
No, This doesn't reflect the latest commit in PR #917.  

Just add
```
#define _POSIX_C_SOURCE 199309L // need clock_gettime()
```
and
```
include <sys/time.h>
```

linux0ne-su, 2020-03-30:
https://github.com/Proxmark/proxmark3/pull/917/commits/9dd8c2af44dc79101c4fc07ced60c38e76e25045#

我是按照上方链接的代码进行更改的。

[image]
[image]

pwpiwi, 2020-03-30:
Green lines have to be added, and red lines to be removed. In this case this reverts the changes made in the first commit of the PR. Select "Files changed" to see the final files after applying all commits:

[image]

Or even better: create a separate branch in your local repository and pull the PR:
```
git checkout -b pwpiwi-fix_get_clocktime master
git pull https://github.com/pwpiwi/proxmark3.git fix_get_clocktime
```

linux0ne-su, 2020-03-30:
Thank you. You've been a great help already.
I will compile the latest code tomorrow evening.

git checkout -b pwpiwi-fix_get_clocktime master
git pull https://github.com/pwpiwi/proxmark3.git fix_get_clocktime

There is no computer by my side now.

If there is still a problem, I'll ask you again.

Thank you so.....

pwpiwi, 2020-03-31:
> If there is still a problem, I'll ask you again.

If this fixes the problem, please tell us too.

pwpiwi, 2020-04-07:
Taking no feedback as positive feedback. Merged #917.

closed 2020-04-07 by pwpiwi

linux0ne-su, 2020-04-08:
非常抱歉，我最近一段工作特别忙，没有及时回复您，抱歉！

1、我回到电脑前，我按照您的方式调整了代码，即将被标注为绿色的代码添加到comms.c文件中，将标注为红色的代码从comms.c文件中删除。

2、但，即使是这样做了之后，还是提示错误，如图所示。

[image]

3、因为中国网络的原因，我们访问github.com网站的时候非常的慢，下载源代码也非常的慢，并且因为我之前更改了已下载好的proxmark3代码，所以在使用git pull --rebase的时候，就显示不成功。也就是说，您如果因为我的提问，而回归代码到最初的状态的话，那因为网络的原因，导致我不能下载这个更新后的源码。

linux0ne-su, 2020-04-08:
[image]

[image]

pwpiwi, 2020-04-08:
There is no need to download all source code again. Just do `git pull` on master branch. It will only download the differences.

linux0ne-su, 2020-04-11:
(base) ooooo@Guo-desktop:~/proxmark3/client$ git pull --rebase
remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 8 (delta 2), reused 6 (delta 2), pack-reused 0
展开对象中: 100% (8/8), 完成.
来自 https://github.com/Proxmark/proxmark3
   5bc3841..c85805b  master     -> origin/master
更新 5bc3841..c85805b
Fast-forward
 CI/.travis.yml | 27 ++++++++++++++++-----------
 client/comms.c |  3 +++
 2 files changed, 19 insertions(+), 11 deletions(-)
当前分支 master 是最新的。
(base) ooooo@Guo-desktop:~/proxmark3/client$ 

------------------------------------------------------------------------------------------------------
make clean  && make
.........
pcsc.c:27:10: fatal error: winscard.h: 没有那个文件或目录
 #include <winscard.h>
          ^~~~~~~~~~~~
compilation terminated.
Makefile:348: recipe for target 'obj/pcsc.o' failed
make[1]: *** [obj/pcsc.o] Error 1
make[1]: 离开目录“/home/ooooo/proxmark3/client”
Makefile:33: recipe for target 'client/all' failed
make: *** [client/all] Error 2
(base) ooooo@Guo-desktop:~/proxmark3$

linux0ne-su, 2020-04-11:
sudo apt-get install libpcsclite-dev

make clean && make
........

make[1]: 离开目录“/home/ooooo/proxmark3/armsrc”
make -C recovery  all
make[1]: 进入目录“/home/ooooo/proxmark3/recovery”
arm-none-eabi-objcopy --gap-fill=0xff --pad-to 0x00102000 -O binary ../bootrom/obj/bootrom.elf bootrom.bin
arm-none-eabi-objcopy --gap-fill=0xff -O binary ../armsrc/obj/fullimage.elf fullimage.bin
cat bootrom.bin fullimage.bin > proxmark3_recovery.bin
make[1]: 离开目录“/home/ooooo/proxmark3/recovery”
make -C tools/mfkey  all
make[1]: 进入目录“/home/ooooo/proxmark3/tools/mfkey”
gcc -std=c99 -D_ISOC99_SOURCE -I../../include -I../../common -I../../client -Wall -O3 -c -o crypto1.o ../../common/crapto1/crypto1.c
gcc -std=c99 -D_ISOC99_SOURCE -I../../include -I../../common -I../../client -Wall -O3 -c -o crapto1.o ../../common/crapto1/crapto1.c
gcc -std=c99 -D_ISOC99_SOURCE -I../../include -I../../common -I../../client -Wall -O3 -c -o parity.o ../../common/parity.c
gcc -std=c99 -D_ISOC99_SOURCE -I../../include -I../../common -I../../client -Wall -O3 -c -o util_posix.o ../../client/util_posix.c
gcc -std=c99 -D_ISOC99_SOURCE -I../../include -I../../common -I../../client -Wall -O3 -c -o mfkey.o ../../client/mifare/mfkey.c
gcc -std=c99 -D_ISOC99_SOURCE -I../../include -I../../common -I../../client -Wall -O3  -o mfkey32 crypto1.o crapto1.o parity.o util_posix.o mfkey.o mfkey32.c
gcc -std=c99 -D_ISOC99_SOURCE -I../../include -I../../common -I../../client -Wall -O3  -o mfkey64 crypto1.o crapto1.o parity.o util_posix.o mfkey.o mfkey64.c
make[1]: 离开目录“/home/ooooo/proxmark3/tools/mfkey”

ok 
done

think you  very very very..........