Proxmark/proxmark3 #925: bash: make: command not found

PhoenixCampbell, 2020-03-27:
Windows, first run, would not allow a "make clean && make all" on step 11. Followed online youtube tutorial as well as written tutorial and ran into the same problem. Did notice in the current ProxSpace master there is no pm3 folder as well (obviously) no bash files that were supposed to be in it. I would love additional help with this as it is for a college class extra credit (Networking and Data Analysis). I have since closed the command window so any errors previously have not shown up upon re-running the batch files. I know that makes it harder to diagnose, but I'm open for help and feedback.

pwpiwi, 2020-03-28:
If the errors do not show up again, what is your issue now?

PhoenixCampbell, 2020-03-28:
My understanding is that you had to do step 11, as it still will not allow me to make clean and make all. It doesnt seem to recognise make as an actual command.

pwpiwi, 2020-03-29:
You need to provide more details.
> Followed online youtube tutorial as well as written tutorial

Which? Link please.
> would not allow a "make clean && make all"

What did you type, what was the response? Screenshots or cut&paste here, please.

PhoenixCampbell, 2020-04-01:
https://www.youtube.com/watch?v=zzF0NCMJnYU

pm3 ~$ make clean && make all                                                                                           
bash: make: command not found

pwpiwi, 2020-04-01:
So you are following the installation instructions on https://github.com/RfidResearchGroup/proxmark3 and if you follow it, you are trying to compile the software from https://github.com/RfidResearchGroup/proxmark3. You therefore should ask for help on https://github.com/RfidResearchGroup/proxmark3 instead of here.

Also, if you are following this instructions, "make clean && make all" is step 9. Step 11 is "cd client".

closed 2020-04-07 by pwpiwi