Proxmark/proxmark3 #927: Can't update Bootloader

thunderbuns, 2020-04-02:
When trying to run the 'pm3-flash-all' command during the initial setup, I encountered the error shown below: 
[image]

Researching this issue led me to [this](http://www.proxmark.org/forum/viewtopic.php?id=6706) page. However, the suggested action to update the bootloader was to run the following command: 'client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf' This command cannot be run as there is no flasher file contained anywhere within the project. As such, the issue lies within the installation instructions and the necessity for them to be updated. It should be noted that I am operating with root privileges and have already run the 'make clean && make all' and the 'make install' commands.

Is there something that I am missing or does the issue lie in the documentation?

iceman1001, 2020-04-02:
This is not a question for the offical repo.  

```
./pm3-flash-bootrom
if, ok,
./pm3-flash-fullimage
```

thunderbuns, 2020-04-03:
For starters, thanks for the speedy response. Your instructions worked perfectly and everything flashed completely fine. As a side note, I was following the guide on the Github page [(link)](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/0_Compilation-Instructions.md). Would it be possible to update this page with these instructions?

iceman1001, 2020-04-03:
The guide is correct as it is.  What would be helpful is a guide when going from offical repo to rrg/iceman repo.  Feel free to contribute.

closed 2020-04-03 by iceman1001