Proxmark/proxmark3 #928: Can't get the stdout of proxmark3.exe immediately when using QProcess

wh201906, 2020-04-07:
I'm trying to write a GUI app by Qt. And now I run into a problem.  
I use QProcess to communicate with the PM3 client, but the client doesn't send me the feedback right after I send a command to it. I can only get the stdout when the buffer is full(after a bunch of commands was sent) or the client is closed.  
If I use the terminal(cmd,bash) to interact with the client, it works fine. And due to the existing GUI for proxmark3, I suspect the issue is caused by Qt(since there are lots of C# based GUI for PM3).  
I found a [page](https://forum.qt.io/topic/11076/get-each-line-of-qprocess-output) relating to this issue.  
I wonder could you plz set the stdout into line buffered explicitly or just disable buffering, which is supposed not to affect other functions so much.  And this could be an enhancement since it makes the PM3 client more easily to interact with other programs.

wh201906, 2020-04-07:
I‘m not a native English speaker, so if anything is ambiguous, plz tell me and I will try to explain it.

pwpiwi, 2020-04-07:
Did you try the `-f` option when starting proxmark3 ?

wh201906, 2020-04-07:
It works with -f option.
Thanks a lot! I didn't notice these options before.

pwpiwi, 2020-04-07:
You are welcome. Looking forward to your GUI.

closed 2020-04-07 by pwpiwi