Proxmark/proxmark3 #933: Signing Proxmark Drivers

DavidBerdik, 2020-04-24:
If I have a signing certificate, is there an easy way for me to use it to sign my builds of the proxmark driver for Windows?

iceman1001, 2020-04-24:
look in the folder https://github.com/Proxmark/proxmark3/tree/master/driver  for a file to sign and try to install to system.

pwpiwi, 2020-04-25:
There should be no need for a Proxmark specific driver any more after Windows 7 being sunset.

DavidBerdik, 2020-05-03:
@iceman1001 Thanks! I have never signed a driver before, so it took me some time to figure out, but I was finally able to successfully do it.

@pwpiwi Even though Windows 7 is not longer supported, I still have a system that uses it that I need to use a Proxmark on, and I wanted to avoid disabling driver signature verification.

closed 2020-05-03 by DavidBerdik

iceman1001, 2020-05-04:
If you wanna share that signed driver, I take a copy :)

DavidBerdik, 2020-05-05:
@iceman1001 Sure! Run the `install-proxmark3-driver.bat` file as an administrator to install it. The driver is signed using an expired certificate, so as part of the installation process, your system clock will temporarily be set back to 2011. Once the install completes, it should be set back.

[Proxmark3 Signed Driver Installer.zip](https://github.com/Proxmark/proxmark3/files/4584166/Proxmark3.Signed.Driver.Installer.zip)

iceman1001, 2020-05-06:
Thanks! Does Win10 allow for outdated drivers to be run?   Interesting.

DavidBerdik, 2020-05-06:
@iceman1001 You're welcome! You can indeed use this on Windows 10! I've recorded a short demo video showing the installation process for both Windows 10 and Windows 7. In both cases, I started by plugging in my proxmark without installing any drivers. The process works the same in both cases, but as you will see in the video, it takes a little longer for the installation to complete on Windows 7.

https://youtu.be/OGZS-2NBBQ0

iceman1001, 2020-05-07:
In the video,  I didn't see turning back the clock.    And yes, that was some serious articfacting.

DavidBerdik, 2020-05-07:
@iceman1001 I have reuploaded the video to fix the artifacting and edited my previous post to include the new video.

As for setting back the clock, watch the video again, but this time, pay attention to the system date when the batch script is executed. You will see that for both Windows 10 and Windows 7, the date is temporarily reverted to September 1, 2011.

Points in the video that demonstrate this:
- Windows 10: https://youtu.be/iDBnggJUsjo?t=28
- Windows 7: https://youtu.be/iDBnggJUsjo?t=97