Proxmark/proxmark3 #934: I want to change the DEZ

BioBeehive, 2020-04-25:
I want to change the DEZ 10 specifically my questions are outlined in my post on the DangerousThings Forum
Any answers or suggestions would be amazing 

https://forum.dangerousthings.com/t/custom-dez-lines-in-an-em410x-tag/6575

mwalker33, 2020-04-25:
Do you want to know what to program the card with such that it will give you a DEZ10 value of X ?

The EM4100 tag ID is made up of 1 Byte Customer ID and 4 Bytes Card ID/Data
I believe the DEZ 10 is the CardID/Data 4 bytes shown in decimal (not hex)

e.g.
```
EM TAG ID      : AA11223344
DEZ 10         : 0287454020
```
DEZ 10 converted to hex :  11223344

so if we dont care what the Customer ID is, but what a DEZ 10 of 123456789
then, convert that to Hex (zero pad for 4 bytes/8 nibbles) :‭ 075BCD15
Prepend the Customer ID (e.g. 00) : 00‬075BCD15
Then write that to the card as the new EM4100 ID

Verify via lf search
```
EM TAG ID      : 00075BCD15
DEZ 10         : 0123456789
```

BioBeehive, 2020-04-25:
thank you so much I finally got the tag to spit out what I wanted 😂

closed 2020-04-25 by BioBeehive