Proxmark/proxmark3 #935: Can't connect PM3 after follow document [Flashing your Proxmark3 to the latest firmware]

MatrixShieldShip, 2020-04-29:
Yesterday I follow the document for windows user - part [Flashing your Proxmark3 to the latest firmware], and today I cannot connect to PM3. The red light C is always turn on. I remember there has't any error during doing those step.

How can I do? Thank you.

pwpiwi, 2020-04-29:
Which operating system are you using?

pwpiwi, 2020-04-29:
What was your previous firmware? Did you flash the bootloader as well?

MatrixShieldShip, 2020-04-30:
I use windows. I don't  know the previous firmware, I'm new to the PM3, it is interesting and I want to learn it. All I did is just follow the document😂.

Can I reset it through USB or I should buy a JLink to solve this problem?

thank you.

pwpiwi, 2020-04-30:
(I thought as well that you are using Windows since you followed the document for windows users.)

* Which Version of Windows are you using?
* Did you see any errors or warnings during step 11 of section "Development environment installation"? If yes, please post them here.
* Do you still see your Proxmark as a Serial Device in Device Manager? On which com Port?

MatrixShieldShip, 2020-04-30:
My Windows version is 10.0.18362.

I didn't see any errors during step 11.

I didn't see Proxmark in the Device manager since I cannot connect it.

pwpiwi, 2020-05-06:
Did it ever work before?

You may have managed to brick your Proxmark. See the debricking sections in the Wiki.

PS: Questions not related to software development are better asked in the forum.

closed 2020-09-09 by pwpiwi