Proxmark/proxmark3 #936: Proxmark Dosent Appear

Mr-Momen, 2020-05-01:
Hey There And Thanks For This Great Project ....
I Have Proxmark3 easy and when i plugd it in my pc all the lights turned on the light [[[A, B, C, D]]]] and when i try to flash the drimware it dosent respond also when i try to see it in the usb and type this command :- dmesg | grep -i usb
i get this :- 
============================================
[ 3939.627071] usb 3-3: new full-speed USB device number 35 using xhci_hcd
[ 3939.755121] usb 3-3: device descriptor read/64, error -71
[ 3942.967004] usb 3-3: new full-speed USB device number 36 using xhci_hcd
[ 3943.095006] usb 3-3: device descriptor read/64, error -71
=============================================

NOTE :- my os is ubuntu

finally i think the problem is all the lights if it is turned on so how to fix that

thanks

Mr-Momen, 2020-05-01:
[image]

pwpiwi, 2020-05-06:
LEDs are on by default and are switched off at the very beginning of the boot process. When they stay on, chances are high that you either managed to brick your Proxmark by destroying the bootloader or that it is physically damaged.

You may have a look at the debricking sections of the Wiki.

PS: better ask questions which are not related to software development on the forum.

closed 2020-05-06 by pwpiwi