Proxmark/proxmark3 #937: Can't find a key for mifare classic 1k

MohammadSecure, 2020-05-09:
i'm working  with **proxmark 3 ** 3 , bootrom: master/v3.1.0-197-gebf1404-suspect 2020-05-05 20:03:22
In my test for a the card: with this syntax: `hf search`   =>> the result is :

> TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
> proprietary non iso14443-4 card found, RATS not supported          
> No chinese magic backdoor command detected          
> Prng detection: WEAK          
> Valid ISO14443A Tag Found - Quiting Search

          
But with use this syntax:
`hf mf chk *1 ? d `   ==>>

> No valid keys found.

pwpiwi, 2020-05-09:
`hf mf chk` without a dictionary only checks for a few default keys. A few more keys (90+) are provided in default_keys.dic. An even bigger key list is available in the RFID repository.

MohammadSecure, 2020-05-10:
> `hf mf chk` without a dictionary only checks for a few default keys. A few more keys (90+) are provided in default_keys.dic. An even bigger key list is available in the RFID repository.

I have checked this attack with syntax ` hf mf chk *1 default_keys.dic` , But the result was this : 
> No valid keys found.

pwpiwi, 2020-05-10:
Obviously none of the keys in default_keys.dic matches any of the keys on your card. This is not a software related issue but an indication that the card publisher is not completely dumb. Please ask any further questions in the forum.

closed 2020-05-10 by pwpiwi

MohammadSecure, 2020-05-10:
Thanks 
Please send for me address of **forum**

pwpiwi, 2020-05-10:
http://www.proxmark.org/forum/index.php

MohammadSecure, 2020-05-12:
> http://www.proxmark.org/forum/index.php

### **Thanks**