Proxmark/proxmark3 #949: Undefined reference to Memcpy

Knocks83, 2020-09-05:
Hello, I'm trying to compile the repo on Arch Linux.
When I do `make install` it gives me this error:
```
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: obj/usb_cdc.o: in function `cmd_send':
usb_cdc.c:(.text+0x6f8): undefined reference to `memcpy'
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: obj/usb_cdc.o: in function `cmd_send_old':
usb_cdc.c:(.text+0x75e): undefined reference to `memcpy'
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: obj/bootrom.o: in function `BootROM':
bootrom.c:(.text+0x38e): undefined reference to `memset'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:41: obj/bootrom.elf] Error 1
make[1]: Leaving directory '/home/knocks/pm3/proxmark3/bootrom'
make: *** [Makefile:29: bootrom/all] Error 2
```
GCC version 10.2.0

pwpiwi, 2020-09-09:
Are you sure that you are trying to compile **this** repo? There is no `memcpy` in usb_cdc.c and no `memset` in bootrom.c

siisko, 2021-01-22:
Hi,

I am compiling **this** repo but the same problem as @Knocks83.

AndColla, 2021-01-27:
Hi,
I also have the same issue compiling this repo

pwpiwi, 2021-01-27:
Confirmed. This is caused by gcc 10 which replaces some loops by calls to memcpy(), memset(), memmove(). Working on it.

AndColla, 2021-01-27:
I fixed it addming `-fno-builtin` in `APP_CFLAGS` in `proxmark3/bootrom/Makefile`
I don't know if it is the right way to to it, but it worked

Einstein2150, 2021-02-09:
same here:

```
brew install proxmark/proxmark3/proxmark3
==> Installing proxmark3 from proxmark/proxmark3
==> Downloading https://github.com/proxmark/proxmark3/archive/v3.1.0.tar.gz
Already downloaded: /Users/a1121460/Library/Caches/Homebrew/downloads/471b24c1bba9caa8a706f5ea1c5a18ba115fc5b69e70495544065fb64fc444e0--proxmark3-3.1.0.tar.gz
==> make clean
==> make all
Last 15 lines from /Users/a1121460/Library/Logs/Homebrew/proxmark3/02.make:
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -Os -I. -mthumb -mthumb-interwork -o obj/version.o version.c 
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -Os -I. -mthumb-interwork -o obj/ram-reset.o ram-reset.s
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -Os -I. -mthumb-interwork -o obj/flash-reset.o flash-reset.s
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -Os -I. -mthumb -mthumb-interwork -o obj/cmd.o ../common/cmd.c 
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -Os -I. -mthumb -mthumb-interwork -o obj/usb_cdc.o ../common/usb_cdc.c 
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -Os -I. -mthumb -mthumb-interwork -o obj/bootrom.o bootrom.c 
arm-none-eabi-gcc -nostartfiles -nodefaultlibs -Wl,-gc-sections -n -Wl,-T,ldscript-flash,-Map,obj/bootrom.map -o obj/bootrom.elf obj/version.o obj/ram-reset.o obj/flash-reset.o obj/cmd.o obj/usb_cdc.o obj/bootrom.o -lgcc
/usr/local/Cellar/arm-none-eabi-gcc/10-2020-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: obj/cmd.o: in function `cmd_send':
cmd.c:(.text+0x3e): undefined reference to `memset'
/usr/local/Cellar/arm-none-eabi-gcc/10-2020-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: cmd.c:(.text+0x66): undefined reference to `memcpy'
/usr/local/Cellar/arm-none-eabi-gcc/10-2020-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: obj/bootrom.o: in function `BootROM':
bootrom.c:(.text+0x3a2): undefined reference to `memset'
collect2: error: ld returned 1 exit status
make[1]: *** [obj/bootrom.elf] Error 1
make: *** [bootrom/all] Error 2
```

YoungJules, 2021-03-16:
Compiling on Windoze 10 was giving me the same error, 'fixed' by adding the suggested -fno-builtin flag as suggested by @AndColla above.

ulixs, 2021-03-28:
Getting the same error on Win10, at:

> APP_CFLAGS = -I.#
added
> APP_CFLAGS = -I-fno-builtin.#

Still not working - not getting bootrom.elf or fullimage.elf

pwpiwi, 2021-03-29:
Did you try with the fixes in PR #969 ?

cynfewl, 2021-03-30:
> Did you try with the fixes in PR #969 ?

This fixed this issue for me. Thank you very much @pwpiwi.

closed 2021-03-30 by pwpiwi