Proxmark/proxmark3 #952: Patch submission to fix compilation update for more recent versions of the cross-compiler tool chain.

mcarey42, 2020-10-13:
commit 002e088f3a68f9a061f205d6c1ebb9aae6bf7c29 (HEAD -> master)
Author: Mark Carey <mcarey@solcore.net>
Date:   Tue Oct 13 12:49:48 2020 -0400

    Removed unused duplicate declaration from armsrc/hitagS.c which was preventing compilation due to duplicate global variable.
    
    Altered common/Makefile.common to use -no-builtin flag on compiler to compiment the -nodefaultlibs at link.  This forces the compiler to generate inline versions for memcpy and memset, which were not found due to the removal of the stdl
ib elements from link phase.

diff --git a/armsrc/hitagS.c b/armsrc/hitagS.c
index 8f40146..b4f9cd2 100644
--- a/armsrc/hitagS.c
+++ b/armsrc/hitagS.c
@@ -42,7 +42,7 @@ static int sof_bits;            //number of start-of-frame bits
 static byte_t pwdh0, pwdl0, pwdl1;  //password bytes
 static uint32_t rnd = 0x74124485;   //randomnumber
 static int test = 0;
-size_t blocknr;
+//size_t blocknr;
 bool end=false;
 
 // Single bit Hitag2 functions:
diff --git a/common/Makefile.common b/common/Makefile.common
index 116a1b2..f1bc40a 100644
--- a/common/Makefile.common
+++ b/common/Makefile.common
@@ -67,10 +67,10 @@ VPATH = . ../common ../common/crapto1 ../common/mbedtls ../fpga ../zlib
 
 INCLUDES = ../include/proxmark3.h ../include/at91sam7s512.h ../include/config_gpio.h ../include/usb_cmd.h $(APP_INCLUDES)
 
-CFLAGS =  -c $(INCLUDE) -Wall -Werror -pedantic -std=c99 $(APP_CFLAGS)
+CFLAGS =  -c $(INCLUDE) -Wall -Werror -pedantic -fno-builtin -std=c99 $(APP_CFLAGS)
 LDFLAGS = -nostartfiles -nodefaultlibs -Wl,-gc-sections -n
 
-LIBS = -lgcc
+LIBS = -lgcc 
 
 THUMBOBJ = $(patsubst %.c,$(OBJDIR)/%.o,$(notdir $(THUMBSRC)))
 THUMBOPTOBJ = $(patsubst %.c,$(OBJDIR)/%.o,$(notdir $(THUMBOPTSRC)))

pwpiwi, 2020-10-14:
Agreed with the removal of `blocknr` in hitagS.c (it isn't used there anyway). Consequentially it should be made `static` (or even better: become an argument to the respective functions) in hitag.c.

Regarding -fno-builtin: -std=c99 already prevents using builtin functions. `memcpy()` and `memset()` are defined in string.c and therefore the stdlib is not required during linking. Which errors do you get during compiling/linking?

libin-ka, 2020-10-14:
proxmark3> hf se

 UID : 20 7e 5c 0b
ATQA : 00 04
 SAK : 28 [1]
TYPE : JCOP31 or JCOP41 v2.3.1
 ATS : 10 78 80 90 02 20 90 00 00 00 00 00 20 7e 5c 0b 6f 9c
       -  TL : length is 16 bytes
       -  T0 : TA1 is present, TB1 is present, TC1 is present, FSCI is 8 (FSC = 256)
       - TA1 : different divisors are NOT supported, DR: [], DS: []
       - TB1 : SFGI = 0 (SFGT = (not needed) 0/fc), FWI = 9 (FWT = 2097152/fc)
       - TC1 : NAD is NOT supported, CID is supported
       -  HB : 20 90 00 00 00 00 00 20 7e 5c 0b
No chinese magic backdoor command detected
Prng detection: WEAK

Valid ISO14443A Tag Found - Quiting Search

proxmark3> hf 14a apdu -S -k 00A40000023F00
>>>>[sel keep ] 00 a4 00 00 02 3f 00
ERROR: Proxmark connection timeout.
proxmark3>

mcarey42, 2020-10-14:
> Agreed with the removal of `blocknr` in hitagS.c (it isn't used there anyway). Consequentially it should be made `static` (or even better: become an argument to the respective functions) in hitag.c.

Logical choices.  I approve (as if that means anything at all :-) ).

> 
> Regarding -fno-builtin: -std=c99 already prevents using builtin functions. `memcpy()` and `memset()` are defined in string.c and therefore the stdlib is not required during linking. Which errors do you get during compiling/linking?

I was getting link errors for strcpy and memset in the bootrom link phase from the common/usb_cdc.c file (I assume it was macro expansion bringing them in).  I was under the impression that setting c99 would resolve that, as well, but it didn't in this case.  When I added the -fno-builtin it cleared right up.  Suspect toolchain problems, but didn't try to trace it down further than building and testing the device.  Tests were all clean and everything worked.  

Exact toolchain version from gcc: 10.2.0 (devkitARM release 55)

Let me know if I can help any further, thanks.

pwpiwi, 2020-10-19:
@libin-ka: please don't hijack other threads and please don't double post.

@mcarey42: please revert your changes, then do `make clean && make all` in the bootrom directory and post the output here. It is still not clear what happens here. Adding -fno_builtin should add more references to be resolved instead of removing unresolved references.

pwpiwi, 2020-11-11:
Probably a duplicate of issue #949

closed 2020-11-11 by pwpiwi