Proxmark/proxmark3 #954: Edit command

MohammadSecure, 2020-11-10:
Hi 
how to change source code for command `hf search`
for example:
**change 52 code to 26 : for start of hf search** 

### proxmark3> hf list 14a
Recorded Activity (TraceLen = 113 bytes)          
          
Start = Start of Frame, End = End of Frame. Src = Source of transfer          
All times are in carrier periods (1/13.56Mhz)          
          
      Start |        End | Src | Data (! denotes parity error, ' denotes short bytes)            | CRC | Annotation         |          
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
          **0 |        992 | Rdr | 52'                                                             |     | WUPA**          
       **7040 |       8032 | Rdr | 52'                                                             |     | WUPA**          
       9140 |      11508 | Tag | 04  00                                                          |     |           
      14080 |      16544 | Rdr | 93  20                                                          |     | ANTICOLL          
      17588 |      23412 | Tag | c5  92  7a  00  2d                                              |     |           
      25856 |      36320 | Rdr | 93  70  c5  92  7a  00  2d  43  6a                              |  ok | SELECT_UID          
      37428 |      40948 | Tag | 08  b6  dd                                                      |     |           
      79872 |      84576 | Rdr | 60  00  f5  7b                                                  |  ok | AUTH-A(0)          
      86324 |      90996 | Tag | d9  3f  2a  a6                                                  |     |           
proxmark3> 
###

pwpiwi, 2020-11-11:
You would need to modify function `GetATQA()`
https://github.com/Proxmark/proxmark3/blob/e6158a48ecc4dbd344604fcb562315931c68d2ff/armsrc/iso14443a.c#L1674
But I have no idea why you would want to do that.

MohammadSecure, 2020-11-23:
my idea is detect two each together cards,
i want have a custom search function that edited `hf search` until a after first run without 
 field interruption again run `hf 14a info` but between two `hf 14a info` , send halt command `0x50` until first card go to idle state. After that run `hf 14a info` for second to detect second card.