Proxmark/proxmark3 #955: MSYS2 is detected as Linux

xobs, 2020-11-19:
I'm using MSYS2, downloaded from https://www.msys2.org/

The Proxmark3 `Makefile` appears to consult `uname` looking for the string `MINGW`. However, under MSYS, this string is `msys`:

```
$ uname
MSYS_NT-10.0-19041

smcro@Pazuzu MSYS /c/Users/smcro/Code/Proxmark/proxmark3
$
```

This results in the `Makefile` attempting to build for Linux, which fails.

xobs, 2020-11-19:
I got it building with the following patch:

```patch
diff --git a/Makefile b/Makefile
index 1ab83f9..54cfc3f 100644
--- a/Makefile
+++ b/Makefile
@@ -7,7 +7,7 @@ ifeq ($(shell echo ""),)
 DELETE=rm -rf
 FLASH_TOOL=client/flasher
 platform=$(shell uname)
-ifneq (,$(findstring MINGW,$(platform)))
+ifneq (,$(findstring MSYS,$(platform)))
 FLASH_PORT=com3
 PATHSEP=\\#
 else
diff --git a/client/Makefile b/client/Makefile
index 2b5e9ae..5ae7d67 100644
--- a/client/Makefile
+++ b/client/Makefile
@@ -42,7 +42,7 @@ endif

 platform = $(shell uname)

-ifneq (,$(findstring MINGW,$(platform)))
+ifneq (,$(findstring MSYS,$(platform)))
        PCSC_INCLUDES :=
        PCSC_LIBS = -lwinscard
 else
@@ -56,7 +56,7 @@ else
 endif

 LUAPLATFORM = generic
-ifneq (,$(findstring MINGW,$(platform)))
+ifneq (,$(findstring MSYS,$(platform)))
                LUAPLATFORM = mingw
 else
        ifeq ($(platform),Darwin)

```

pwpiwi, 2020-11-19:
Mingw is the recommended environment. Your patch would break building under Mingw.

closed 2020-11-19 by pwpiwi