Proxmark/proxmark3 #956: ProxmarkPro hw status/hf search is not working: unknown command: 0x0108

Darkkey, 2020-11-20:
Hello, I'm trying to do `hw status` and `hf search` on my ProxmarkPro and getting `unknown command:: 0x0108` error. 

Here is the short log on what's happening:
```
~/tools/proxmark3/client$ sudo ./proxmark3 /dev/ttyACM0
Prox/RFID mark3 RFID instrument          
bootrom: pmpro-1.1 2019-09-26 15:16:46
os: pmpro-1.1 2019-09-26 15:16:47
FPGA image built for 3s250evq100 on 2018/11/09 at 16:40:49
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 180840 bytes (34%). Free: 343448 bytes (66%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> hw status
#db# unknown command:: 0x0108          
Status command failed. USB Speed Test timed out       
```

Next, I'm trying to apply any HF card, e.g. Milfare Classic 4K
```   
proxmark3> hf search
#db# unknown command:: 0x0108                
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
Card (MIM 0) read, use 'hf legic decode' or          
'data hexsamples 0' to view results          

Valid Legic Tag Found - Quiting Search
          
proxmark3> data hexsamples 0
00 00 00 00 00 00 00 00          
proxmark3> hf legic decode
Sending bytes to proxmark failed          
^C
```

So in USB-connected mode it's not detecting any cards, while in the independent mode everything (HF/LF) works perfectly. Could it be some firmware incompatibility with current version of client?

pwpiwi, 2020-11-20:
> Could it be some firmware incompatibility with current version of client?

Absolutely. The Client Software and the Proxmark Firmware need to match. Please note that ProxmarkPro is not supported by this repository.

closed 2020-11-20 by pwpiwi

Darkkey, 2020-11-20:
@pwpiwi What's the official repository for ProxmarkPro then? Official website points to the client software here. (e.g. https://store.ryscc.com/blogs/news/proxmarkpro-client-setup-getting-started )

Darkkey, 2020-11-20:
Nevermind, apparently they enforce to use their closed-source prebuilt binaries of Proxmark3 client.

pwpiwi, 2021-01-14:
The ProxmarkPro source code is here: https://cdn.shopify.com/s/files/1/0847/7088/files/pmpro-1.1.tgz