Proxmark/proxmark3 #961: #winscard.h missing

anonanonanonanon, 2020-12-15:
How do i fix this issue i have found a simmilar problem with solution but that fix unfortunately didnt work out for me

pwpiwi, 2021-01-14:
Which operating system are you using and with which source file do you get this error? Please paste the output of your make command here.

fsota, 2021-01-23:
I have the same issue. With Parrot OS:
pcsc.c:27:10: fatal error: winscard.h: No such file or directory
   27 | #include <winscard.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:348: obj/pcsc.o] Error 1
make[1]: Leaving directory '/home/user/proxmark3/client'
make: *** [Makefile:33: client/all] Error 2
┌─[✗]─[user@parrot-virtual]─[~/proxmark3]

fsota, 2021-01-23:
[image]

fsota, 2021-01-23:
working after install libpcsclite-dev
$ apt install libpcsclite-dev

pwpiwi, 2021-01-24:
Well, as documented in the Wiki (https://github.com/Proxmark/proxmark3/wiki/Ubuntu-Linux#setup-and-build-for-ubuntu)

closed 2021-01-24 by pwpiwi