Proxmark/proxmark3 #967: lf em 410xwrite not working

AddaxSoft, 2021-03-17:
I'm on kali linux:
```
Linux megakali 5.9.0-kali1-amd64 #1 SMP Debian 5.9.1-1kali2 (2020-10-29) x86_64 GNU/Linux
```
I did `killall modem_manager`
and run the proxmark3:

```
Prox/RFID mark3 RFID instrument
bootrom: master/v3.1.0-204-g555fa19-suspect 2021-03-16 21:55:47
os: master/v3.1.0-204-g555fa19-suspect 2021-03-16 21:55:47
fpga_lf.bit built for 2s30vq100 on 2019/11/21 at 09:02:37
fpga_hf.bit built for 2s30vq100 on 2020/03/05 at 19:09:39
SmartCard Slot: not available

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 207484 bytes (40%). Free: 316804 bytes (60%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
```

I did `lf search u` to identify my tag
```
Checking for known tags:

EM410x pattern found:

EM TAG ID      : 010EB39AFD

Possible de-scramble patterns
Unique TAG ID  : 8070CD59BF
HoneyWell IdentKey {
DEZ 8          : 11770621
DEZ 10         : 0246651645
DEZ 5.5        : 03763.39677
DEZ 3.5A       : 001.39677
DEZ 3.5B       : 014.39677
DEZ 3.5C       : 179.39677
DEZ 14/IK2     : 00004541618941
DEZ 15/IK3     : 000551648319935
DEZ 20/ZK      : 08000700121305091115
}
Other          : 39677_179_11770621
Pattern Paxton : 29874429 [0x1C7D8FD]
Pattern 1      : 8024025 [0x7A6FD9]
Pattern Sebury : 39677 51 3382013  [0x9AFD 0x33 0x339AFD]

Valid EM410x ID Found!
```

Ok, so it's a `EM410x` tag. I read the UID with this command:
```
proxmark3> lf em 410xread 1
#db# EM TAG ID: 010eb3xxxx - (39677_179_1177xxxx)
#db# Stopped
```

Now I want to write the UID to another tag:
```
proxmark3> lf em 410xwrite 010eb23xxxx 1
Writing T55x7 tag with UID 0x010eb2f720 (clock rate: 64)
#db# Started writing T55x7 tag ...
#db# Clock rate: 64
```

nothing happens next
even `hw tune` doesn't response anymore:

```
roxmark3> lf em 410xwrite 010eb2xxxx 1
Writing T55x7 tag with UID 0x010ebxxxx (clock rate: 64)
#db# Started writing T55x7 tag ...
#db# Clock rate: 64
proxmark3> hw tune

Measuring antenna characteristics, please wait...........
No response from Proxmark. Aborting...
```

I noticed an error message in my dmesg:
```
cdc_acm 1-1.2:1.0: failed to set dtr/rts
```

now I can't read or do anything until I exit. 
When I read after exit the tag is still old value.

so far I tried:
* multiple clock rate
* different type t55x7, t55555

sometimes I get this:
```
proxmark3> #db# Tag T55x7 written with 0xff8060edcbe79418
```

but again that's not true, the UID is not written. What could be the issue?

AddaxSoft, 2021-03-17:
I made some progress:

I learned that some tags are configured as read only, so finally I was able to write to a writable tag.
However, I have a writable tag that I could clone with a cheap device from aliexpress, but cannot write to it with the proxmark3 easy.

I can't figure out why

mwalker33, 2021-03-17:
```I have a writable tag that I could clone with a cheap device from aliexpress, but cannot write to it with the proxmark3 easy```
Sounds like the cloner put a password on it.  If so, you will need the password to make changes.

AddaxSoft, 2021-03-19:
not really, simply because when I use the cloner to write on the t5577 card I can still over-write on it with my proxmark3 easy.

mwalker33, 2021-03-20:
OK, the tag that you can write to with the cloner, but not the proxmark, does the proxmark find it with an 

lf t55 det

AddaxSoft, 2021-03-27:
I'm replacing hardware JUST IN CASE, but I remember I was able to detect the tag with `lf search`

iceman1001, 2021-03-27:
You can always try out the RRG/Iceman repo and see if that one works better.   Worth a try before you replace your hw

AddaxSoft, 2021-04-07:
ok got new hardware, but same issue persists. I'm happy to debug and test.
`lf t55 det` returns this:

```
proxmark3> lf t55 read
Reading Page 0:
blk | hex data | binary
----+----------+---------------------------------
```

mwalker33, 2021-04-12:
That shows a `lf t55 read`, can you try the `lf t55 detect`