Proxmark/proxmark3 #973: hf mfu commands fail for NATG213

Bloup1net, 2021-04-12:
I can't dump a NTAG213 contents with hf commands, they keep failing with :

WARNING: iso14443a card select failed         

I have a proxmark3:

proxmark3> hw version
Prox/RFID mark3 RFID instrument          
bootrom: master/v3.1.0-209-g6116334-suspect 2021-04-06 14:33:11
os: master/v3.1.0-209-g6116334-suspect 2021-04-06 14:33:03
fpga_lf.bit built for 2s30vq100 on 2019/11/21 at 09:02:37
fpga_hf.bit built for 2s30vq100 on 2020/03/05 at 19:09:39
SmartCard Slot: available
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 207558 bytes (40%). Free: 316730 bytes (60%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory   

The hf search result for my NTAG213 tag result:

proxmark3> hf search
          
 UID : 04 c8 35 92 aa 5e 80          
ATQA : 00 44          
 SAK : 00 [2]          
WARNING: iso14443a card select failed          
TYPE : MIFARE Ultralight (MF0ICU1) <magic>          
MANUFACTURER : NXP Semiconductors Germany          
proprietary non iso14443-4 card found, RATS not supported          
No chinese magic backdoor command detected          

Valid ISO14443A Tag Found - Quiting Search
          
Some hf mfu commands that fail:

proxmark3> hf mfu dump
WARNING: iso14443a card select failed   

proxmark3> hf mfu info
WARNING: iso14443a card select failed

MKesenheimer, 2022-10-01:
Slightly lift the tag (1-2cm from the antenna) and try again.
This helped in my case.

JediYarosh, 2023-03-07:
I have exactly the same issue with exactly the same output, except the UID of course. When I slightly move the chip away from the antenna I do not get anything. I need to basically touch it in a certain position to read it.

joef12345, 2023-04-10:
i'm having the same problem with a mifare classic card too.