Proxmark/proxmark3 #981: flasher : Error: Unexpected reply 0x00fe (expected ACK)

ThePirateWhoSmellsOfSunflowers, 2021-06-30:
Hi,

I have the following error when I try to flash my proxmark with the last commit height.

```
$ sudo client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf 
Loading ELF file 'bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000cd4->0x00000cd4) [R X] @0x298

Waiting for Proxmark to appear on /dev/ttyACM0 .
 Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to appear on /dev/ttyACM0 ......
 Found.

Flashing...
Writing segments for file: bootrom/obj/bootrom.elf
 0x00100000..0x001001ff [0x200 / 1 blocks]Error: Unexpected reply 0x00fe (expected ACK)
 ERROR
Error writing block 0 of 1
```

Result of the `hw version` command.
```
proxmark3> hw version
Prox/RFID mark3 RFID instrument          
bootrom: master/v3.0.1-459-gdc3e2ac-suspect 2018-10-18 08:36:37
os: master/v3.0.1-459-gdc3e2ac-suspect 2018-10-18 09:01:24
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2018/09/12 at 15:18:46
SmartCard Slot: not available
          
uC: AT91SAM7S256 Rev A          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes. Used: 197873 bytes (75%). Free: 64271 bytes (25%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 256K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> exit
```

Thanks for the help !

:sunflower:

ThePirateWhoSmellsOfSunflowers, 2021-07-12:
Hey!

I solved this issue by flashing the proxmark with a BusPirate and now I can flash it with `flasher`. Howerver I'm still interested if someone know the root cause of the error. Is it something related with the `Lock Error Bit` (I had to `flash protect 0 0 15 off` during the JTAG flashing) ?

:sunflower:

pwpiwi, 2021-07-12:
> Unexpected reply 0x00fe (expected ACK)

0x00fe means NACK. This response is sent when trying to flash to an invalid address, when the programming keyword is wrong, or when lock bits are set. The former two "should never happen", so your assumption about the lock bits should be correct.

ThePirateWhoSmellsOfSunflowers, 2021-07-12:
I have flashed the proxmark multiple times before the issue, I don't know why this time lock bits were set. Howerver, thanks for your response.

:sunflower:

closed 2021-07-12 by ThePirateWhoSmellsOfSunflowers