Proxmark/proxmark3 #983: Waiting for a response from the proxmark...

droidandy, 2021-07-20:
Hi I have a proxmark rdv4 and macos

I managed to run a few commands but i needed to update the firmware to give me access to `hf mf ekeyprn d`.

I went through the wiki guide here. made and flashed the bootloader and fullimage.elf

```
sudo ./client/flasher /dev/tty.usbmodem142301 armsrc/obj/fullimage.elf
Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x00030510->0x00030510) [R X] @0x94
1: V 0x00200000 P 0x00132510 (0x000012f8->0x000012f8) [RW ] @0x305a4
Note: Extending previous segment from 0x30510 to 0x31808 bytes

Waiting for Proxmark to appear on /dev/tty.usbmodem142301 .
 Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to appear on /dev/tty.usbmodem142301 .
 Found.

Flashing...
Writing segments for file: armsrc/obj/fullimage.elf
 0x00102000..0x00133807 [0x31808 / 397 blocks]............................................................................................................................................................................................................................................................................................................................................................................................................. OK

Resetting hardware...
All done.

Have a nice day!
```

however now when i connect to the proxmark i get

```
proxmark3 /dev/cu.usbmodem142301                                      
proxmark3> hf search
          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button 
```

nothing i do seems to fix this, i can no longer run any commands. 

can you please help?

pwpiwi, 2021-07-22:
When you are coming from a pretty old firmware your bootloader might be outdated as well. Try flashing a new bootloader. In the current (inconsistent) state you probably need to press and hold the proxmark button during the flashing process. And be aware that the Proxmark may appear under a different port now.

APOFISAN, 2023-03-26:
and how is it done?

martin-eberle, 2024-10-23:
I have a similar problem. Neither flashing a new boot loader nor flashing a new firmware works. In both cases it's stuck at `Waiting for a response from the proxmark...`.

Example output:

`sudo proxmark3-flasher /dev/tty.usbmodem143401 -b /usr/local/share/firmware/bootrom.elf
Loading ELF file '/usr/local/share/firmware/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000cd4->0x00000cd4) [R X] @0x298

Waiting for Proxmark to appear on /dev/tty.usbmodem143401 .
 Found.
Waiting for a response from the proxmark...
You can cancel this operation by pressing the pm3 button`

Tried also kali linux but that doesn't help.

How can this be fixed?

martin-eberle, 2024-10-23:
I just used the fork https://github.com/RfidResearchGroup/proxmark3 instead and now it works!