Proxmark/proxmark3 #985: cloning keyfob 13.56 MHz

JOJOLEE1981, 2021-08-19:
I got an error message as following when I tried to clone the keyfob 13.56 MHz

[usb] pm3 --> hf iclass dump --ki 0
[+] Using AA1 (debit) key[0] AE A6 84 A6 DA B2 32 78
**[=] Card has at least 2 application areas. AA1 limit 18 (0x12) AA2 limit 31 (0x1F)**
.
[!!] failed to communicate with card
[usb] pm3 -->                                                                       

Please see the ver of the PM3

[usb] pm3 --> hw ver

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.13441-725-g49ccd6ae1 2021-08-17 10:46:50
  compiled with MinGW-w64 10.3.0 OS:Windows (64b) ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.13441-725-g49ccd6ae1 2021-08-17 10:46:37
       os: RRG/Iceman/master/v4.13441-725-g49ccd6ae1 2021-08-17 10:46:45
  compiled with GCC 10.1.0

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 53% used )

[usb] pm3 -->                                            

Any help with the iclass cloning instruction will be great. Thanks

iceman1001, 2021-08-19:
Wrong repo.

pwpiwi, 2021-08-19:
You are obviously using the RRG software. Please post your issues in the respective github repository https://github.com/RfidResearchGroup/proxmark3

closed 2021-08-19 by pwpiwi

JOJOLEE1981, 2021-08-20:
> You are obviously using the RRG software. Please post your issues in the respective github repository https://github.com/RfidResearchGroup/proxmark3

Hi pwpiwi 

please advise where I should post. Thanks

merlokk, 2021-08-20:
http://www.proxmark.org/  -- forum or discord
https://github.com/RfidResearchGroup/proxmark3/blob/master/README.md