Proxmark/proxmark3 #994: proxmark3 usb modem not showing up after flashing

toothbrushes, 2022-01-28:
```
System: MacBook Pro (16-inch, 2021) Apple M1 Max (arm64)
Proxmark3 version 3.1.0
```

After installing `proxmark3` through the homebrew tap, I tried to use my device but I was getting errors when running `lf sniff` (closed the terminal, so I can't share what they were).  Though, `data tune` was working fine.

So I thought perhaps my software version had a mismatch with the hardware version, and decided to flash my hardware following these instructions:

https://github.com/Proxmark/homebrew-proxmark3

```shell
[~] brew tap proxmark/proxmark3
[~] brew install proxmark3
[~] proxmark3-flasher /dev/tty.usbmodem21201 /opt/homebrew/share/firmware/fullimage.elf
Loading ELF file '/opt/homebrew/share/firmware/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002c808->0x0002c808) [R X] @0x94
1: V 0x00200000 P 0x0012e808 (0x00001a60->0x00001a60) [RW ] @0x2c89c
Note: Extending previous segment from 0x2c808 to 0x2e268 bytes

Waiting for Proxmark to appear on /dev/tty.usbmodem21201 .
 Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to appear on /dev/tty.usbmodem21201 ..
 Found.

Flashing...
Writing segments for file: /opt/homebrew/share/firmware/fullimage.elf
 0x00102000..0x00130267 [0x2e268 / 370 blocks].................................................................................................................................................................................................................................................................................................................................................................................. OK

Resetting hardware...
All done.

Have a nice day!
```

Now my usbmodem is no longer showing up.  😭

There is a green & a orange LED that are on, and will turn off and back on about every 15 seconds.

Not quite sure what to do at this point.  I watch a video and someone said it won't show up because there's a mismatch now between the fullimage & bootloader, and said to unplug the device, then plug it backs a mismatch.

toothbrushes, 2022-01-28:
Ugh it seems like the device may be bricked because I should be using this other `proxmark3` https://github.com/RfidResearchGroup/proxmark3

And also it needs to be compiled for `PLATFORM=PM3EASY`

https://github.com/RfidResearchGroup/proxmark3/issues/217

toothbrushes, 2022-01-28:
Wait, that's not correct either, I believe it is

```Makefile
PLATFORM=PM3GENERIC
PLATFORM_SIZE=512
```

🤷 

I suppose I'll give https://github.com/RfidResearchGroup/proxmark3 a try when the new device arrives, and it may just work without needing flashing since it seems this library isn't being used by the hardware anymore 👀

pwpiwi, 2022-01-31:
First of all, don't panic. Flashing fullimage to your device was successful, so it is very unlikely that it is bricked. You didn't mention your firmware release before the flashing but I would assume that the bootloader now no longer matches the fullimage.

Try flashing the bootloader. You will need to press the Proxmark Button all the times from powering up until flashing is complete.