Proxmark/proxmark3 #995: Build error on Fedora 35

bfnserra, 2022-01-31:
Hello Proxmark team ! 

I'm trying to build proxmark on fedora 35 (arm-none-eabi-gcc 11.1.0) and got the following error:

```
make[1]: Entering directory '/home/nico/applications/proxmark3/bootrom'
perl ../tools/mkversion.pl .. > version.c || cp ../common/default_version.c version.c
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -I. -Os -mthumb -mthumb-interwork -o obj/version.o version.c 
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -I. -Os -mthumb-interwork -o obj/ram-reset.o ram-reset.s
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -I. -Os -mthumb-interwork -o obj/flash-reset.o flash-reset.s
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -I. -Os -mthumb-interwork -o obj/string.o string.c 
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -I. -Os -mthumb -mthumb-interwork -o obj/usb_cdc.o ../common/usb_cdc.c 
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -I. -Os -mthumb -mthumb-interwork -o obj/bootrom.o bootrom.c 
bootrom.c: In function 'UsbPacketReceived':
bootrom.c:122:20: error: array subscript 'uint32_t {aka long unsigned int}[0]' is partly outside array bounds of 'char[1]' [-Werror=array-bounds]
  122 |           flash_mem[i] = c->d.asDwords[i];
      |           ~~~~~~~~~^~~
bootrom.c:22:43: note: while referencing '_flash_start'
   22 | extern char _bootrom_start, _bootrom_end, _flash_start, _flash_end;
      |                                           ^~~~~~~~~~~~
bootrom.c:114:10: error: array subscript 'uint32_t {aka volatile long unsigned int}[0]' is partly outside array bounds of 'char[1]' [-Werror=array-bounds]
  114 |         p[i+arg0] = c->d.asDwords[i];
      |         ~^~~~~~~~
bootrom.c:22:43: note: while referencing '_flash_start'
   22 | extern char _bootrom_start, _bootrom_end, _flash_start, _flash_end;
      |                                           ^~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [../common/Makefile.common:82: obj/bootrom.o] Error 1
make[1]: Leaving directory '/home/nico/applications/proxmark3/bootrom'
make: *** [Makefile:29: bootrom/all] Error 2
```

Here is  information about arm-none-eabi-gcc:

```
~ » arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/arm-none-eabi/11.1.0/lto-wrapper
Target: arm-none-eabi
Configured with: ../gcc-11.1.0/configure --prefix=/usr --mandir=/usr/share/man --with-pkgversion='Fedora 11.1.0-2.fc35' --with-bugurl=https://bugzilla.redhat.com/ --infodir=/usr/share/info --target=arm-none-eabi --enable-interwork --enable-multilib --with-python-dir=share/arm-none-eabi/gcc-11.1.0/python --with-multilib-list=rmprofile --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --enable-languages=c,c++ --with-newlib --disable-nls --disable-shared --disable-threads --with-gnu-as --with-gnu-ld --with-gmp --with-mpfr --with-mpc --with-headers=yes --with-system-zlib --with-sysroot=/usr/arm-none-eabi
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 11.1.0 (Fedora 11.1.0-2.fc35) 
```

Thanks in advance for your help !

aldemarf, 2022-02-04:
Same problem here. [Manjaro / kernel 5.16]

```
❯ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/11.2.0/lto-wrapper
Target: arm-none-eabi
Configured with: /build/arm-none-eabi-gcc/src/gcc-11.2.0/configure --target=arm-none-eabi --prefix=/usr --with-sysroot=/usr/arm-none-eabi --with-native-system-header-dir=/include --libexecdir=/usr/lib --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-system-zlib --with-newlib --with-headers=/usr/arm-none-eabi/include --with-python-dir=share/gcc-arm-none-eabi --with-gmp --with-mpfr --with-mpc --with-isl --with-libelf --enable-gnu-indirect-function --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='Arch Repository' --with-bugurl=https://bugs.archlinux.org/ --with-multilib-list=rmprofile
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 11.2.0 (Arch Repository)
```

pwpiwi, 2022-02-08:
How did you manage to come that far? I am already stuck at compiling version.c:
```
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -I. -Os -mthumb -mthumb-interwork -o obj/version.o version.c 
In file included from ../include/usb_cmd.h:21,
                 from ../include/proxmark3.h:17,
                 from version.c:1:
/usr/lib/gcc/arm-none-eabi/11.1.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
    9 | # include_next <stdint.h>
      |                ^~~~~~~~~~
compilation terminated.
```

bfnserra, 2022-02-09:
@pwpiwi - looks like you have a different error, probably related to some dependencies... ? 
I googled "stdint.h: No such file or directory" and found some interesting things, hope it helps !

pwpiwi, 2022-02-10:
@bnfserra: `sudo dnf install arm-none-eabi-newlib` fixed my issue on Fedora. I now can confirm your issue. Looking into it..

bfnserra, 2022-03-31:
In the meantime, I was able to start playing via docker : `docker run -it --device="/dev/ttyACM0" --privileged -v /tmp/share:/opt/proxmark3/Data iceman1001/proxmark3:1.7.0`
Hope it helps.

crazyquark, 2022-10-07:
I made these changes to have it compile:
https://github.com/Proxmark/proxmark3/commit/ee8491b04a5ef3950c2fdc1dd2c2a14706247e7e

AndreKR, 2022-10-29:
Same issue (`array subscript 'uint32_t {aka long unsigned int}[0]' is partly outside array bounds of 'char[1]'`) on Windows following [these instructions](https://github.com/Proxmark/proxmark3/wiki/Windows#development-environment-installation).

iceman1001, 2022-10-29:
you can also try https://github.com/rfidresearchgroup/proxmark3   which is a more up-to-date client / firmware for your proxmark3

AndreKR, 2022-10-29:
The changes from @crazyquark's commit solved the issue.

nyetwurk, 2023-05-15:
This is not the way you're supposed to use linker defined symbols.
It should be
```
extern uint8_t _bootrom_start[], _bootrom_end[], _flash_start[], _flash_end[];
```

nyetwurk, 2023-05-15:
https://github.com/rfidresearchgroup/proxmark3 has
```
extern uint32_t _bootrom_start[], _bootrom_end[], _flash_start[], _flash_end[], _osimage_entry[];
```
that makes a ton more sense. thanks.