Proxmark/proxmark3 #1001: Malicious URLs on Wiki

ayancey, 2022-06-18:
Hey,

It looks like multiple people are persistently changing the URL on the wiki page for Windows installation from ProxSpace to malicious crypto trading bots.

https://github.com/Proxmark/proxmark3/wiki/Windows

Someone changed the URL from https://github.com/Gator96100/ProxSpace/archive/master.zip to `https://github.com/lampii/VideoCaptureUtility/releases/download/42/master.zip`

The user lampii has already been terminated or deleted but the download link still works.

pwpiwi, 2022-06-19:
@0xFFFF (owner of this repository): Would it make sense to allow Wiki changes to contributors only?

ayancey, 2022-06-20:
Hello everyone.

ayancey, 2022-06-20:
FYI, this continues to be an issue. The attacker has gone and tried to revert the change I made, re-enabling the malicious link. https://github.com/Proxmark/proxmark3/wiki/Windows/_history

AkechiShiro, 2022-06-20:
Some contributors need to step in ASAP, this is a joke that someone is toying with a security researcher project like this

0xFFFF, 2022-06-21:
In response to this particular issue I have enabled:
- `General` / `Require contributors to sign off on web-based commits`
- `General` / `Features` / `Wikis` / `Restrict editing to users in teams with push access only`

Please inform me (just as @pwpiwi has done) if additional changes are required.

ayancey, 2022-06-21:
Thank you. If anyone downloaded and ran that executable, I would highly recommend you nuke your system. The attacker got access to my password manager, and GitHub account while I was away from my computer. I opened an abuse case with GitHub to get the account terminated.

[image]

closed 2022-06-24 by ayancey