Proxmark/proxmark3 #1004: Windows build error in bootrom.c

hakeila, 2022-11-07:
Hi,

I have been trying to build proxmark3 project under Windows 10 but I am getting the following syntax errors in bootrom.c (it is mainly array out of bound error)

```
make[1]: Leaving directory '/pm3/client'
make -C bootrom  all
make[1]: Entering directory '/pm3/bootrom'
perl ../tools/mkversion.pl .. > version.c || cp ../common/default_version.c version.c
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -I. -Os -mthumb -mthumb-interwork -o obj/version.o version.c
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -I. -Os -mthumb-interwork -o obj/ram-reset.o ram-reset.s
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -I. -Os -mthumb-interwork -o obj/flash-reset.o flash-reset.s
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -I. -Os -mthumb-interwork -o obj/string.o string.c
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -I. -Os -mthumb -mthumb-interwork -o obj/usb_cdc.o ../common/usb_cdc.c
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -std=c99 -I. -Os -mthumb -mthumb-interwork -o obj/bootrom.o bootrom.c
bootrom.c: In function 'UsbPacketReceived':
bootrom.c:122:20: error: array subscript 'uint32_t {aka long unsigned int}[0]' is partly outside array bounds of 'char[1]' [-Werror=array-bounds]
  122 |           flash_mem[i] = c->d.asDwords[i];
      |           ~~~~~~~~~^~~
bootrom.c:22:43: note: object '_flash_start' of size 1
   22 | extern char _bootrom_start, _bootrom_end, _flash_start, _flash_end;
      |                                           ^~~~~~~~~~~~
bootrom.c:114:10: error: array subscript 'uint32_t {aka volatile long unsigned int}[0]' is partly outside array bounds of 'char[1]' [-Werror=array-bounds]
  114 |         p[i+arg0] = c->d.asDwords[i];
      |         ~^~~~~~~~
bootrom.c:22:43: note: object '_flash_start' of size 1
   22 | extern char _bootrom_start, _bootrom_end, _flash_start, _flash_end;
      |                                           ^~~~~~~~~~~~**
cc1.exe: all warnings being treated as errors
make[1]: *** [../common/Makefile.common:82: obj/bootrom.o] Error 1
```
make[1]: Leaving directory '/pm3/bootrom'
make: *** [Makefile:29: bootrom/all] Error 2

chopinwong01, 2022-11-30:
I got exactly the same error...

hakeila, 2022-12-01:
Hi @chopinwong01,

It seems that I was trying to install an incorrect version of proxmark3. Here is the github repo that I tried and successfully built it on my widows machine.

https://github.com/RfidResearchGroup/proxmark3

chopinwong01, 2022-12-01:
> 
Hi @hakeila .

I tried the iceman build with the PM3-generic platform setting, all works fine. 
Thanks, mate.

hakeila, 2022-12-01:
@chopinwong01 glad to know that it worked for you. 

You are most welcomed :)

Peter-x000x, 2023-03-28:
FYI:
The changes from [@crazyquark's commit](https://github.com/Proxmark/proxmark3/issues/995#issuecomment-1271277447) solved the issue.
=> [ee8491b](https://github.com/Proxmark/proxmark3/commit/ee8491b04a5ef3950c2fdc1dd2c2a14706247e7e)

meditant, 2023-04-13:
Same probleme

acasanova99, 2024-02-25:
This is not only a windows issue, it happens on Linux as well. It is solved just by modifying a couple of lines in the code as @Peter-x000x has mentioned. Ty

ZHJ0125, 2024-05-15:
> FYI: The changes from [@crazyquark's commit](https://github.com/Proxmark/proxmark3/issues/995#issuecomment-1271277447) solved the issue. => [ee8491b](https://github.com/Proxmark/proxmark3/commit/ee8491b04a5ef3950c2fdc1dd2c2a14706247e7e)

It's worked for me.

adrianvaszul, 2025-02-11:
> FYI: The changes from [@crazyquark's commit](https://github.com/Proxmark/proxmark3/issues/995#issuecomment-1271277447) solved the issue. => [ee8491b](https://github.com/Proxmark/proxmark3/commit/ee8491b04a5ef3950c2fdc1dd2c2a14706247e7e)

It's working! It's working!