Proxmark/proxmark3 #1007: stuck in update firmware( Received packet OLD frame with payload too short)

reliveinfire, 2023-05-05:
Hi community,
    after 'make PLATFORM=PM3GENERIC all' and pm3-flash-all

i have one log
[!] ⚠️  Received packet OLD frame with payload too short? 22/534
[=][=] You can cancel this operation by pressing the pm3 button

forever, is there way to fix this?
    
```
    root@relive-VirtualBox:/home/relive/project/proxmark3# ./pm3-flash-all
[=] Session log /root/.proxmark3/logs/log_20230505.txt
[+] About to use the following files:
[+]    /home/relive/project/proxmark3/client/../bootrom/obj/bootrom.elf
[+]    /home/relive/project/proxmark3/client/../armsrc/obj/fullimage.elf
[+] Loading ELF file /home/relive/project/proxmark3/client/../bootrom/obj/bootrom.elf
[+] ELF file version Iceman/master/v4.16191-333-g52981476e 2023-05-05 18:24:25 24e01a23a

[+] Loading ELF file /home/relive/project/proxmark3/client/../armsrc/obj/fullimage.elf
[+] ELF file version Iceman/master/v4.16191-333-g52981476e 2023-05-05 18:28:08 24e01a23a

[+] Waiting for Proxmark3 to appear on /dev/ttyACM0
 🕑  59 found
[!] ⚠️  Received packet OLD frame with payload too short? 22/534
[=] You can cancel this operation by pressing the pm3 button
```

thanks

reliveinfire, 2023-05-05:
try  ./pm3-flash-bootrom
and got below message,
 is there any way to fix this?
```
root@relive-VirtualBox:/home/relive/project/proxmark3# ./pm3-flash-bootrom
[=] Session log /root/.proxmark3/logs/log_20230505.txt
[+] About to use the following file:
[+]    /home/relive/project/proxmark3/client/../bootrom/obj/bootrom.elf
[+] Loading ELF file /home/relive/project/proxmark3/client/../bootrom/obj/bootrom.elf
[+] ELF file version Iceman/master/v4.16191-333-g52981476e 2023-05-05 18:24:25 24e01a23a

[+] Waiting for Proxmark3 to appear on /dev/ttyACM0
 🕑  59 found
[+] Entering bootloader...
[+] (Press and release the button only to abort)
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0
 🕖  44 found
[=] You can cancel this operation by pressing the pm3 button

[!] ⚠️  Communicating with Proxmark3 device failed
```

thanks

iceman1001, 2023-05-05:
you sure you are posting this in the right repo?
https://github.com/rfidresearchgroup/proxmark3

reliveinfire, 2023-05-05:
opps, thanks for reply,
it seems that it's my environment problems, i have ubuntu runs in my virtual box 
and i am lazy to follow wiki to standard instructions, i bind the window usb into virtual box.
this is not working. i reinstall mingw64, and it works now.
my bad.

closed 2023-05-05 by reliveinfire