Proxmark/proxmark3 #1008: autopwn store incorrect key value when hardnested fail

reliveinfire, 2023-05-05:
Hi sir,
    in some caes, the hf mf hardnested may fail to get the correct keys
in autopwn, it stores the incorrect key value
(i've use hf mf hardnested with script to find all correct keys)

case 1
```
[=]       28 |    1883 | Apply Sum(a8) and all bytes bitflip properties          |             nan |  nand
[+] target sector   0 key type B -- found valid key [ 000000000000 ]
```

correct one
```
[=]       35 |    2930 | Brute force phase completed.  Key found: 8642289466E5   |               0 |    0s
[+] target sector   2 key type A -- found valid key [ 8642289466E5 ]
```
but next fail case, it stores previous correct key
```
[=]       63 |    2210 | Apply Sum(a8) and all bytes bitflip properties          |             nan |  nand
[+] target sector   2 key type B -- found valid key [ 8642289466E5 ]

[=]       26 |    1662 | Apply Sum(a8) and all bytes bitflip properties          |             nan |  nand
[+] target sector   3 key type A -- found valid key [ 8642289466E5 ]

[=]       26 |    1662 | Apply Sum(a8) and all bytes bitflip properties          |             nan |  nand
[+] target sector   3 key type A -- found valid key [ 8642289466E5 ]

[=]       26 |    1662 | Apply Sum(a8) and all bytes bitflip properties          |             nan |  nand
[+] target sector   3 key type A -- found valid key [ 8642289466E5 ]
```
for nan cases, the fail found keys are supposed to store with nothing or special symbol.

is there better way to auto retry hf mf hardnest? i wrote one lua script to do this, but i dont know how to intercept the log.
i just try every valid key many times and use terminal find to check if proxmark found the key.

thanks

reliveinfire, 2023-05-05:
i did not finish the full test, this card is full encryption,
maybe at the last step the autopwn checks the keys and re-start to find the incorrect key, is this right?

pwpiwi, 2023-05-05:
https://github.com/Proxmark/proxmark3/issues/1007#issuecomment-1536067663

reliveinfire, 2023-05-06:
after trace the code, the autopwn did not re run the hardnest for the incorrect key,
the problems seem that found_key varible did not handle correctly in mfnestedhard  
1. if the brute_force value is 0, 
2. the correct key may be dropped in previous scan, the brute_force function can not find the correct key from the pending  keys

maybe we can add re-try mechanism or add retry parameter in hardnest command to recover this card type
i did this with script which issue hf md hardnest command. and it has chance to recover the correct key.

also, i found that test_grab.py in experimental_lib can parse the terminal log, it would be helpful to auto update the key when brute force run hf md hardnest in script.

closed 2023-05-06 by reliveinfire