Proxmark/proxmark3 #1012: Software stop simulation

alansbor, 2023-08-28:
Hi all!
There is a need to programmatically stop the simulation, because Proxmark 3 easy does not have the ability to change blocks in the recorded memory of the emulator or is it?

There is this version, it works under windows 10

[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.14434-1980-ge10c70adc 2023-03-30 18:44:02 2d0b0eabf
  compiled with............. MinGW-w64 11.2.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: Iceman/master/v4.14434-1980-ge10c70adc 2023-03-30 18:43:24 2d0b0eabf
       os: Iceman/master/v4.14434-1980-ge10c70adc 2023-03-30 18:43:29 2d0b0eabf
  compiled with GCC 10.1.0

 [ FPGA ]
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 60% used )
-----------------------------------------------------------------

I am trying to emulate a Mifare Ultralight card
and i need to change some blocks in simulator memory or change data in json file and reload it for simulation.

Is it possible to execute the command
hf mfu wrbl for data located in the memory of the simulator, I load the json file

 local new_json = json.encode(json_data)
  local fname = string.format("%04X.json", i) 
  local f = io.open(fname, "w")
  f:write(new_json)
  f:close()
  
 core.console("hf mfu eload -f " .. fname)
and run the simulation
 core.console("hf mfu sim -t 2 --uid 04CF698A014A80")
 
but because I can not change the data - write to the emulated memory, I change the data in the json file and turn on the simulation again.
how to programmatically stop simulation or write modified blocks to simulator memory?

=] MIFARE Ultralight override, will use 29 blocks ( 116 bytes )
[=] Uploading to emulator memory
[=] ..............................
[?] You are ready to simulate. See `hf mfu sim -h`
[=] Done!
[?] Try `hf mfu sim -t 7` to simulate an Amiibo.
[+] Emulating ISO/IEC 14443 type A tag with 7 byte UID (04 CF 69 8A 01 4A 80 )
[=] Press pm3-button to abort simulation
[#] Emulator stopped. Trace length: 2659

how to disable simulation abort function [=] Press pm3-button to abort simulation
I need to disable the simulation programmatically.

DjamikTea, 2024-01-25:
Hello, use this `hf mf eclr`

iceman1001, 2024-01-25:
You are in the wrong repo given the firmware you seem to be using.