Proxmark/proxmark3 #1014: Issue cloning ISO 15693 tags

Ytosko, 2023-10-15:
##Please help! Urgent!

Hello there, I am a new user of proxmark3 rdv2 I have got 512k version and flashed latest firmware using brew on Mac OS. I need your help!

I have an ISO 15693 tag and I want to dump and clone it to simulate the tag. Here the things that I have done so far:

Step 1: *hf search*

Step 2: "*hf 15 dump*"

Then by using the UID I was able to dump 16 blocks of data then I ran "*hf eload*" to load the dump and "hf sim" to simulate the uid and also checked the simulation with my flipper zero which was fine but although the signal wasn't picking by the reader. Then I ran "*hf info*" command and get this result:

```
[+]  UID: E0 04 01 18 01 F3 D4 19

[+] TYPE: NXP(Philips); IC NTP53x2/NTP5210/NTA5332(NTAG 5)

[+] Using UID... E0 04 01 18 01 F3 D4 19

[=] --- Tag Information ---------------------------

[+]       TYPE: NXP(Philips); IC NTP53x2/NTP5210/NTA5332(NTAG 5)

[+]        UID: E0 04 01 18 01 F3 D4 19

[+]    SYSINFO: 00 0F 19 D4 F3 01 18 01 04 E0 00 00 3F 03 01 

[+]      - DSFID supported        [0x00]

[+]      - AFI   supported        [0x00]

[+]      - IC reference supported [0x01]

[+]      - Tag provides info on memory layout (vendor dependent)

[+]            4 (or 3) bytes/blocks x 64 blocks

[=] 

[=] --- NXP Sysinfo

[=]   raw... 00 11 30 00 FF 75 07 04 

[=]     Password protection configuration:

[=]       * Page L read not password protected

[=]       * Page L write not password protected

[=]       * Page H read password protected

[=]       * Page H write password protected

[=]     Lock bits:

[=]       * AFI not locked

[=]       * EAS not locked

[=]       * DSFID not locked

[=]       * Password protection configuration not locked

[=]     Features:

[=]       * User memory password protection supported

[=]       * Counter feature supported

[=]       * EAS ID supported by EAS ALARM command

[=]       * EAS password protection supported

[=]       * AFI password protection supported

[=]       * Extended mode supported by INVENTORY READ command

[=]       * EAS selection supported by extended mode in INVENTORY READ command

[=]       * READ SIGNATURE command supported

[=]       * Password protection for READ SIGNATURE command not supported

[=]       * STAY QUIET PERSISTENT command supported

[=]       * ENABLE PRIVACY command supported

[=]       * DESTROY command supported

[=]       * Additional 32 bits feature flags are not transmitted

[=] 

[=]   EAS (Electronic Article Surveillance) is not active

[=] --- Tag Signature

[=]  IC signature public key name: NXP ICODE DNA, ICODE SLIX2

[=] IC signature public key value:
 048878A2A2D3EEC336B4F261A082BD71F9BE11C4E2E896648B32EFA59CEA6E59F0

[=]     Elliptic curve parameters: NID_secp128r1

[=]              TAG IC Signature: 0107403AD7ECAB5261D71B934DD74F1C315F0D87E40F58B98B64D8911194E6E3

[+]        Signature verification: successful

[=]                   Params used: UID and signature, plain
```

Here I found may be from block 17 to 64 probably *page H* is password protected. Now here are my questions:

1. How can I remove the password or unlock those blocks
2. How can I simulate the tag

Please mention the steps and command which can help me please 🙏

Wombat63, 2024-01-01:
Any progress made? I am facing exactly the same issue - just with RDV4.

iceman1001, 2024-01-01:
This issue has nothing to do with the source code. 

If you have user related questions ,  use the discord server to ask. 

_closing_

closed 2024-01-01 by iceman1001

Amoshajari, 2025-10-16:
@Ytosko Hi were you be able to solve your issue?